stm32duino/STM32LowPower #3: Low power modes

Babyla071, 2018-09-13:
Hi! I've to use low power modes based on the charge of an external battery with stm32L053R8.
I read the voltage of battery through ADC and if this is less of "x" I enter into low power mode.
I readed datasheet and I see five power modes and I want to use low power run and sleep mode. In this library I see  other modes (idle mode, sleep mode , deep sleep mode and shutdown mode). They are the same or not? I can use this for my project? Thank you for help :)

fpistm, 2018-09-13:
Idle = sleep mode with PWR_MAINREGULATOR_ON
sleep = sleep mode with PWR_LOWPOWERREGULATOR_ON
deepSleep = stop mode with PWR_LOWPOWERREGULATOR_ON, PWR_STOPENTRY_WFI
shutdown = standby mode ( board reset when leaves this mode)

Babyla071, 2018-09-13:
The "low-power run mode" supported by the Nucleo board STM32L053R8, which is it exactly?Perhaps this mode can not be achieved through STM32duino.

fpistm, 2018-09-13:
With this library the "low-power run mode" is not available, API is based on Arduino one.
Maybe, a new API could be added to support it for compatible MCU.

To get it the clock config should use the MSI clock source.
Then Low Power Run Mode could be call to go in  LP RUN and the set Regulator is in LP mode .

fpistm, 2018-09-19:
As a first read, why doing a `digitalWrite(USER_BTN,LOW);`, it's an input.
The attachInterrupt is to be able to wakeup the board on an external event but only SYS Wakeup pins can be used.
For the L053:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_L053R8/PinNamesVar.h
```
#ifdef PWR_WAKEUP_PIN1
    SYS_WKUP1 = PA_0,
#endif
#ifdef PWR_WAKEUP_PIN2
    SYS_WKUP2 = PC_13,
#endif
```
So USER_BTN can be used as it is a SYS WKUP pin:
https://github.com/stm32duino/Arduino_Core_STM32/blob/0f1a0ec7eb26fe369a4d5e3354c77d24026ca902/variants/NUCLEO_L053R8/variant.h#L115
```
// On-board user button
#define USER_BTN PC13
```
So when you press the user button, the board is woken up and the attached handler increase the number.
This is just to know how many time the board exit from sleep mode.

Babyla071, 2018-09-19:
Perfect, I did not remember this SYS wakeup mode for the pins but I don't understand how can I wakes up the board when the value of ADC is highest of "900". Maybe I have to set this SYS wake up pin high with digitalWrite where the condition of the else is true ? (valore>900).
Thank you for the help!

fpistm, 2019-03-07:
Sorry, @Babyla071, I thought I've answered you.
About the ADC value, I guess the simplest way is to go in low power then wake up periodically using alarm to check the read the ADC because in LP mode you will not be able to read ADC.
Or if you want use  the syswakeup, I guess you need external circuit to handle set to high if you reach your value.

fpistm, 2019-05-30:
Closed as no feedback and considered as answered.

closed 2019-05-30 by fpistm