stm32duino/STM32LowPower #6: wakeup from deep sleep with RTC or pin

andraz213, 2019-04-05:
Hey, I want to use deep sleep in my project.
I want to use RTC and pin interrupt to wake up from deep sleep.
I want it to wake up from the thing that happens the first. 
Is that even possible?
(I can manage it on my own, I just want to know if that's possible)

fpistm, 2019-04-05:
I think you can using the current API.
attachInterruptWakeup and enableWakeupFrom

andraz213, 2019-04-05:
Okay thank you.
I was trying to do it like attachinterrputwakeup and deepsleep(30000) and it didn't work. 
Then I tried this code and it worked :D

`#include "STM32LowPower.h"
#include <STM32RTC.h>

STM32RTC& rtc = STM32RTC::getInstance();

static uint32_t atime = 1;
static byte seconds = 0;
static byte minutes = 0;
static byte hours = 0;

static byte weekDay = 1;
static byte day = 1;
static byte month = 1;
static byte year = 18;

void setup() {
  pinMode(PA10, INPUT_PULLUP);
  rtc.begin();
  rtc.setTime(hours, minutes, seconds);
  rtc.setDate(weekDay, day, month, year);

  LowPower.begin();
  LowPower.attachInterruptWakeup(PA10, PINhndl, FALLING);
  LowPower.enableWakeupFrom(&rtc, RTChndl, &atime);
}

void loop() {

  delay(5000);
  rtc.setAlarmEpoch( rtc.getEpoch() + 15 );
  LowPower.deepSleep();
}

void RTChndl(void *data) {

}

void PINhndl() {

}`

closed 2019-04-05 by andraz213