stm32duino/STM32LowPower #7: what timer/interupt library to match

OlliHenri, 2019-04-28:
Hello,
thanks for this great lib.
What timer / interupt library matches to this lib for STM32F103C8 / Blue Pill.

are there any conflicts to be expected?

thanks Oliver

fpistm, 2019-04-29:
Hi @OlliHenri 
what did you expect exactly?

OlliHenri, 2019-04-29:
Hello,
I used arduino IDE with STM32GENERIC-master and "HardwareTimer.h" to control my program flow.
Now I like to use low-power-modes and I liked you lib.

But STM32GENERIC-master is not compatible with your library, and I did not find any similar comfortale lib like yours for GENERICSTM32-master.

So I started looking for a timer library to use in combination with STM32LowPower but did not find any...

Do you have any example how to use STM32LowPower in combination with timers?
Are there any conflicts?

thanks -Oliver-

fpistm, 2019-04-29:
Currently, there is no dedicated library to handle timer. There is an issue open for this (https://github.com/stm32duino/Arduino_Core_STM32/issues/146).
You can have a look to huawei fork of STM32GENERIC, I know it back port this lib.
Other way is to backport the HardwareTimer.h fromSTM32GENERIC.

OlliHenri, 2019-04-29:
Hi,
I think backport the HardwareTimer.h fromSTM32GENERIC to Aduino_Core_STM32 is not that easy, HardwarTimer is based on HAL system and as I saw there is no HAL in STM32GENERIC.

How about porting STM32LowPower to STM32GENERIC?
Or any solution more user friendly to switch to LowPower deepsleep / stop mode? and continue program flow from returnpoint.
The combination of user friendly timer/interrupt lib in STM32GENERIC and STM32LowPower enables us to use all the power of STM32 with the used current less than AVR.
My project is to use BluePill as engine room MCU to send sensor readings to central ESP32 and Can-Bus NMEA. It is crucial to use as low power as possible, like on any batterie run projects.

I was looking to do that but ended up void...

-Oliver-

fpistm, 2019-04-29:
STM32GENERIC also use HAL:
https://github.com/danieleff/STM32GENERIC/tree/master/STM32/system/STM32F1

OlliHenri, 2019-04-29:
Hello, sorry in my last post was a mistake, STM32GENERIC and Arduino_STM32-master was mixed up.

How about porting STM32LowPower to Arduino_STM32-master, what is based on libmaple?
As I see there is no HAL in it?
Or any solution more user friendly to switch to LowPower deepsleep / stop mode? and continue program flow from returnpoint.

 I tried also to get the following code to work with Arduino_STM32-master, it compiles but hangs somwhere...

void goToSleep(SleepMode mode) {
    
	// enable the PWR control clock
	//RCC_BASE->APB1ENR |= RCC_APB1ENR_PWREN:
	
	// Clear PDDS and LPDS bits
      PWR_BASE->CR &= ~PWR_CR_PDDS;
      PWR_BASE->CR &= ~PWR_CR_LPDS;
      SCB_BASE->SCR &= ~SCB_SCR_SLEEPONEXIT;

    // Clear previous wakeup register wake up flag
    PWR_BASE->CR |= PWR_CR_CWUF;	// setting CWUF-flag will clear WUF wake up flag by hardware

    if (mode == STANDBY) {
      PWR_BASE->CR |= PWR_CR_PDDS;      // PowerDownDeepSleep	0= enter stopr mode, 1=  
                                                                 //  enter standby mode
	  PWR_BASE->CSR |= PWR_CSR_EWUP;		// enable wakeup pin
    }

    //PWR_BASE->CR |= PWR_CR_LPDS;			// LowPowerDepsleep, 0= powerregulator is on, 1= powerregulator is off

    SCB_BASE->SCR |= SCB_SCR_SLEEPDEEP;			// SCR = system controll register
	//SCB_BASE->SCR |= SCB_SCR_SLEEPONEXIT;		// 0= do not sleep on return from ISR, 
											// 1= go back to sleep when return from ISR
	
	// enable wake up pin
	PWR_BASE->CSR |= PWR_CSR_EWUP;				// enable wakeup pin

    // Now go into stop mode, wake up on interrupt
    asm("    wfi");  				// WFI = wait for interrupt
}

But this also did not work out with Arduino_STM32-master

The combination of user friendly timer/interrupt lib in Arduino_STM32-master and STM32LowPower enables us to use all the power of STM32 with the used current less than AVR.
My project is to use BluePill as engine room MCU on a sailing boat to send sensor readings to central ESP32 and Can-Bus NMEA. It is crucial to use as low power as possible, like on any batterie run projects.

I was looking to do that but ended up void...

-Oliver-

fpistm, 2019-04-29:
Ahhhh 😅
You use Arduino_STM32. This library is for Arduino_Core_STM32 which is HAL based.

OlliHenri, 2019-04-29:
Hi, yes I tried to use Arduino with Blue Pill STM32F103C8 but there are many different cores and all got their issues.
I confirmed that this lib STM32LowPower is working with Arduino_Core_STM32 and was happy with it, but I could not find any working timer / interrupt lib or tutoria siutablel. I found timer/interrupt nicely working with libmaple based core but there the lowPower will not work also not handish by juggling registers.
quite frustrating and I think I will go on with Teensy3.6 but its much more expensive and is quite overpowered for my project. Or maybe I step back to Arduino Pro mini because of LowPower and its well working without overpowered hardware.
-Oliver-

fpistm, 2019-04-30:
Timer are available thanks: https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/timer.h#L217
or thanks HAL (all are accessible at sketch level).
I know this is not easy to use, this will be easier when HardwareTimer will be done.

fpistm, 2019-04-30:
Timer are available thanks: https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/timer.h#L217
or thanks HAL (all are accessible at sketch level).
I know this is not easy to use, this will be easier when HardwareTimer will be done.

A PR has been merge to open the timer access.
See https://github.com/stm32duino/Arduino_Core_STM32/pull/447#issuecomment-464872520
This is merged thanks https://github.com/stm32duino/Arduino_Core_STM32/pull/473 and available in the master

fpistm, 2019-05-23:
This issue relies to stm32duino/Arduino_Core_STM32#146.
Close this one.

closed 2019-05-23 by fpistm