stm32duino/STM32LowPower #8: Pin alternate function

patricklaf, 2019-05-02:
Hello,
I try to use timer input capture with a Nucleo-L476RG, TIM3, channel 1.
The channel input pin is PA6. I linked PA6 to PA0 while PA0 is running a PWM output signal.
Timer is counting, but there is no edge detection.
What is the function to set up PA6 so it will act as timer capture input?
pinMode function doesn't accept alternate as mode parameter.
I tried this but without success:
```
// PA6
pinMode(PA6, INPUT);
pin_function(digitalPinToPinName(PA6), STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0));
```
pinMode call is to setup pin as configured, pin_function call is to set alternate configuration.
Thanks for your help,
Patrick

closed 2019-05-03 by patricklaf

patricklaf, 2019-05-03:
I closed this issue because everything  is working now. Bad wiring, my fault.
Although , it could be interesting to modify the pinMode function to directly handle alternate.
By the way, wrong repository. I followed issues about timer from Arduino_Core_STM32 repository and didn't mind that I was in STM32LowPower repository.
My apologies.

fpistm, 2019-05-03:
No worries