stm32duino/STM32LowPower #10: LowPower.deepSleep doesnt work

Narceen, 2019-08-08:
Hello,

i use an Custom Stm32L432KC Board but if i use LowPower.deepSleep in my Code in Arduino Windows tell me the USB Device got some Error -> Invalid Device Description.

If i use LowPower.sleep, it wont work at all, Serial.print occour infinity no delay at all.

So maybe u can help me and tell what i do wrong?

fpistm, 2019-08-19:
Hi,

I guess you probably need to handle properly USB disconnection or config.
Library does not manages peripherals configurations so before enter in the LP mode you have to manage them.
You can also write your own `SystemClock_ConfigFromStop` which by default call the `SystemClock_Config`.

Narceen, 2019-08-22:
Hi, thx for the replay. I just fixed the usb error with new driver update.
Can u give me a hint how to configurate peripherals for LP Mode?
Thx in advantage

fpistm, 2020-02-05:
OP issue solved.
For the hint, please use the forum for primary support.

closed 2020-02-05 by fpistm