stm32duino/STM32LowPower #11: library cant upload to stm32f103c8

Naveenkato, 2019-09-27:
when i upload this library to stm32f103c8t6 board,I stuck with this error"#error "PWR configuration is missing. Check flag HAL_PWR_MODULE_ENABLED in variants/board_name/stm32yzxx_hal_conf.h"

fpistm, 2019-09-27:
Hi  @Naveenkato 
This library is for this core:
https://github.com/stm32duino/Arduino_Core_STM32

I guess you do not use this core but this one: https://github.com/rogerclarkmelbourne/Arduino_STM32

That's why you get this error as this library is not compatible with it.

Naveenkato, 2019-09-27:
Thank you.I solved this problem.

closed 2019-09-27 by fpistm

jos3c0, 2019-11-01:
@Naveenkato, how do you resolve the problem? Can you share us that info, i have the same issue

fpistm, 2019-11-01:
@jos3c0 
As I explained you probably do not use this core:
https://github.com/stm32duino/Arduino_Core_STM32
but this one:
https://github.com/rogerclarkmelbourne/Arduino_STM32
So, this library is not compatible.

jos3c0, 2019-11-01:
@fpistm 
I am a little confused. I was checking on the issue appearing on
"controller consumes high power during sleep mode #12" and it seems @Naveenkato is using the library with the blue pill(sm32f103c8t6). I want to understand how he managed to do it.

fpistm, 2019-11-01:
I guess he has moved to this core https://github.com/stm32duino/Arduino_Core_STM32