stm32duino/STM32LowPower #12: controller consumes high power during sleep mode

Naveenkato, 2019-09-28:
Hi,
  I'm using stm32f103c8 controller bluepill kit,the controller still consumes 10milliAmps during sleep mode even after i remove the led.

fpistm, 2019-09-28:
Hi, you have to disable all peripherals used.

Naveenkato, 2019-09-28:
Thank you for your response. I have used the "ExternalWakeup" code provided in the examples. I didn't connect any device with the gpio. Just bluepill alone. Here is the code which i have used.

`
#include "STM32LowPower.h"

volatile int repetitions = 1;

// Pin used to trigger a wakeup
#define USER_BTN PA0
const int pin = USER_BTN;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(pin, INPUT_PULLUP);

  LowPower.begin();
  // Attach a wakeup interrupt on pin, calling repetitionsIncrease when the device is woken up
  LowPower.attachInterruptWakeup(pin, repetitionsIncrease, RISING);
}

void loop() {
  for (int i = 0; i < repetitions; i++) {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
  }
  LowPower.sleep();
}

void repetitionsIncrease() {
  repetitions ++;
}`

fpistm, 2019-09-28:
@Naveenkato

looking deeply, the `LowPower.sleep()` on STM32F1 simply a request to WFI:

```C
/**
  * @brief Enters Sleep mode.
  * @note  In Sleep mode, all I/O pins keep the same state as in Run mode.
  * @param Regulator: Regulator state as no effect in SLEEP mode -  allows to support portability from legacy software
  * @param SLEEPEntry: Specifies if SLEEP mode is entered with WFI or WFE instruction.
  *           When WFI entry is used, tick interrupt have to be disabled if not desired as
  *           the interrupt wake up source.
  *           This parameter can be one of the following values:
  *            @arg PWR_SLEEPENTRY_WFI: enter SLEEP mode with WFI instruction
  *            @arg PWR_SLEEPENTRY_WFE: enter SLEEP mode with WFE instruction
  * @retval None
  */
void HAL_PWR_EnterSLEEPMode(uint32_t Regulator, uint8_t SLEEPEntry)
{
  /* Check the parameters */
  /* No check on Regulator because parameter not used in SLEEP mode */
  /* Prevent unused argument(s) compilation warning */
  UNUSED(Regulator);

  assert_param(IS_PWR_SLEEP_ENTRY(SLEEPEntry));

  /* Clear SLEEPDEEP bit of Cortex System Control Register */
  CLEAR_BIT(SCB->SCR, ((uint32_t)SCB_SCR_SLEEPDEEP_Msk));

  /* Select SLEEP mode entry -------------------------------------------------*/
  if(SLEEPEntry == PWR_SLEEPENTRY_WFI)
  {
    /* Request Wait For Interrupt */
    __WFI();
  }
  else
  {
    /* Request Wait For Event */
    __SEV();
    __WFE();
    __WFE();
  }
}
```

Refering to datasheet:

> • Sleep mode
> In Sleep mode, only the CPU is stopped. All peripherals continue.

and here maximum current consumption:

[image]

BP is configured to run at 72 MHz with HSE.
So, it seems 10 mA is normal on this case. Probably some clock are enabled and could be disabled, for example if you enable USB CDC, then the clock is enabled and can be disabled.

Naveenkato, 2019-09-30:
Thank you for your response.i need to work the controller in standby mode,is it possible?

Naveenkato, 2019-10-01:
Thank you for the reply.i need to run the controller in standby mode,In datasheet the current consumption in standby mode is 2uA,Now i using the LowPower.shutdown(),but the current consumption is still 10mA.

Naveenkato, 2019-10-01:
Thank you for the reply.i need to run the controller in standby mode,In datasheet the current consumption in standby mode is 2uA,Now i using the LowPower.shutdown(),but the current consumption is still 10mA.This is the code which i used.

`
#include "STM32LowPower.h"

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Configure low power
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.shutdown(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.shutdown(1000);
}
`

fpistm, 2020-02-13:
Firstly, 
You code is not correct, calling `shutdown()` will put the F103 in standby mode:
[image]
This means after the delay, your board will simply reset so you will never see the led blink (just a quick flash).

I don't know how you measure the consumption of the BP.
I use a Nucleo and have access to IDD. I've tested and get around 3 µA on the STM32F1 series which is as expected. If you measure with USB then I guess this is normal in your case or there is something connected to the MCU which consume.

I close as it works as expected .

closed 2020-02-13 by fpistm