stm32duino/STM32LowPower #13: Library uses 12K of flash memory!

Vitalhb, 2019-10-13:
I am using a STM32F070F6 in my project, with only 32K of flash, but just the low power library added 12K to my code. Is there an easy way to use the STM32 low power modes but using less flash memory?

thanks

fpistm, 2019-10-14:
Hi @Vitalhb 
Right, after doing some check, it seems the use of `mktime` in the STM32RTC (required by the STM32LowPower library) add several code from the libc nano and gcc.
But this is not the main flash consumer.
I'm currently working on decreasing flash size of the core by removing use of `whole-archive`  option.
Anyway even with this I've succeed to build for F030 by doing this:
* Remove of whole-archive
* Serial disabled
* Remove mktime cal in the STM32RTC library
* enable LTO in the optimize menu

> Sketch uses 11836 bytes (72%) of program storage space. Maximum is 16384 bytes.
> Global variables use 776 bytes (18%) of dynamic memory, leaving 3320 bytes for local variables. Maximum is 4096 bytes.

the only thing I could try to do is to see if it is possible to remove `mktime` usage in STM32RTC library.

Vitalhb, 2019-10-14:
Hi Frederic,
Thanks for the quick response. Since I cannot remove Serial (I use a GPS module) and LTO is already enabled, the only option is to remove mktime from the STM32RTC library. I have tested this (just replaced "return mktime(&tm)" by a "return 0" in function getEpoch()) and code size was reduced about 6K. 
Is it possible to calculate the epoch without using mktime?

thanks.

fpistm, 2019-10-14:
I don't know.
Probably this could be done.
Do not hesitate to provide one if you find one.

Vitalhb, 2019-10-24:
What I did was to remove everything I did not need from your libraries STM32LowPower and STM32RTC and replace mktime(&tm) by direct calculations using hours, minutes and seconds to determine the alarm time. 
The simplified library now uses 3K of flash.

mcer12, 2019-11-08:
@Vitalhb Can you share your version of the library?

mcer12, 2019-11-08:
I have tested enabling LTO but the board seems unresponsive after upload even just with the blink sketch.

Vitalhb, 2019-11-08:
> @Vitalhb Can you share your version of the library?

Yes. It is attached here. I have only implemented the deep sleep mode.
[STM32SimpleLowPower.zip](https://github.com/stm32duino/STM32LowPower/files/3826278/STM32SimpleLowPower.zip)

closed 2019-11-08 by Vitalhb

mcer12, 2019-11-08:
I see you also include STM32SimpleRTC.h in the example, can you share it as well?

mcer12, 2019-11-08:
Never mind, my mistake it works perfectly, thanks a ton for this @Vitalhb !

PeeJay, 2020-09-09:
I found some more code savings - rtc.c from the STM32 rtc library uses log2 which is a floating point function. It can be replaced with an integer version which saves around 10k of code.

//#include <math.h>

static inline int log2i(int x) {
    return sizeof(int) * 8 - __builtin_clz(x) - 1;
}

(stolen from https://stackoverflow.com/questions/994593/how-to-do-an-integer-log2-in-c)

fpistm, 2020-09-09:
Thanks @PeeJay 
Do not hesitate to provide a PR for this 😉

ppotter100, 2022-04-15:
I tried this library which seems to work. I only see a lot of drift already for deep sleeps of 60 seconds. Is this normal? 

```

    while(1){
      maintenanceSerial.println("Tick!");
      delay(60);
      STM32SimpleRTC::begin();
      STM32SimpleLowPower::deepSleep(60);
      maintenanceSerial.println("Tock!");
      delay(60);

    }

```

Results in:
```

[2022-04-15 14:55:55.881] Tick!
[2022-04-15 14:56:48.923] Tock!
[2022-04-15 14:56:48.971] Tick!
[2022-04-15 14:57:42.016] Tock!
```