stm32duino/STM32LowPower #16: Unable to upload since TimedWakeup example

gjt211, 2020-02-20:
Hi all, thanks for this library.

I have been playing with the blink example just trying different delays for the LED and programming several times successfully.

Next I uploaded the TimedWakeup example which seems to be working well.
I have now tried to upload another example and I now constantly get the following error.
```
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.3.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 061D08002A12354D314B4E00
ST-LINK FW  : V2J31S7
Voltage     : 3.14V
Error: ST-LINK error (DEV_TARGET_CMD_ERR)
```

I have tried resetting the board, unplugging and replugging the ST-Link.
I am using the blue pill board.

Is there something I need to do to make this work again? I was thinking that now the MCU is in deep sleep mode the programmer is not working but I am unsure. Any help is appreciated.

fpistm, 2020-02-20:
Hi @gjt211 
firstly which board you used ?
The TimeWakeup example use deepSleep which means the MCU is in standby mode.
Then I guess the MCU does not respond to STLINK.
Try to maintain reset before the STM32CubeProgrammer try to get the control the release it.

gjt211, 2020-02-20:
Hi @fpistm and thanks kindly for your response.

I am using the ever popular and not so amazing blue bill.
I believe the problem was with the device being in deep sleep mode, but what was unusual was that I could reprogram it several times (me testing different deep sleep times) successfully but then it just stopped. I assumed the STLINK would bring it out of sleep to program it but it seems I am incorrect (and assuming things is never a good thing to do).

I ended up moving one of the boot jumpers and can now program it again.
I wanted to start moving one of my projects (uses SAMD21G) to the STM32 and need low power. The STM32 devices are substantially less expensive however it is new to me and I need to learn a lot more.
Closing as solved thanks.

closed 2020-02-20 by gjt211

fpistm, 2020-02-21:
In fact,  by default the Reset mode used is software and I guess  in case of deep sleep it is not usable. In that case hard reset shold be used