stm32duino/STM32LowPower #18: Clarification on what functions map to what modes

Eqqman, 2020-03-25:
The STM32L071 has 5 low-power modes: Low-power Run, Sleep, Low-power Sleep, Stop, and Standby.  The STM32LowPower library only has 4 modes: Idle, Sleep, DeepSleep, and Shutdown.  The code's comments are not consistent in describing what STM mode corresponds to what library mode, leading to confusion on which should be used.  I had thought I had found bugs in the library code until I realized it was the comments that were incorrect which had led me to invoke the wrong functions.  

In STM32LowPower.cpp:

Line 63 says that Idle = STM Sleep
Line 77 says that Sleep = STM Sleep, conflicting with Idle
Line 91 says that DeepSleep = STM Stop, but this is not correct, looking at the implementation DeepSleep expects to exit through a SystemWakeup pin, which is the main feature of Standby mode
Line 105 says that Shutdown = STM Standby, likely the only thing in these comments to be correct

Line 128 says that Idle = STM Sleep and Sleep = STM Stop, conflicting with the prior description of Stop
Line 131 says that DeepSleep = STM Standby, which conflicts with the prior description of DeepSleep but agrees with the actual implementation of the function

If the comments could be revised to clarify these issues it would be appreciated.  Note that none of these descriptions claim to invoke Low-power Run or Low-Power sleep, so these descriptions link 4 Arduino functions to only 3 of the 5 STM low-power functions.

fpistm, 2020-03-26:
https://github.com/stm32duino/STM32LowPower/blob/master/README.md#hardware-state

fpistm, 2020-03-26:
As a first note, this library is based on the [Arduino Low Power](https://github.com/arduino-libraries/ArduinoLowPower) one for compatibility and thus match the API naming.
So, each STM32 low power mode has been mapped to the Arduino one and the naming is not exactly the same.

> https://github.com/stm32duino/STM32LowPower/blob/master/README.md#hardware-state

No conflict, this is really STM32 sleep. The main difference is which regulator is used but both called the `HAL_PWR_EnterSLEEPMode`.

> Line 91 says that DeepSleep = STM Stop
> Line 105 says that Shutdown = STM Standby

Same here Arduino API named DeepSleep  called `HAL_PWR_EnterSTOPMode` ans Shutdown the 
`HAL_PWR_EnterSTANDBYMode`.

> Line 128 says that Idle = STM Sleep and Sleep = STM Stop, conflicting with the prior description of Stop
> Line 131 says that DeepSleep = STM Standby, which conflicts with the prior description of DeepSleep

Yes it seems there is a shift. Do not hesitate to provide a PR to fix that.

Eqqman, 2020-03-26:
Yes, no way to give perfect mapping if the Arduino library and the STM32 don't have the same number of low-power modes.  This is just a request to fix the inconsistencies in the comments on how they are mapped.

commit 9fd1535c (2020-03-26): LowPower comment mapping clarifications

Fixes #18
closed 2020-03-26 by fpistm