stm32duino/STM32LowPower #19: LowPower.attachInterruptWakeup() configures external interrupt pins incorrectly

Eqqman, 2020-03-25:
Pins will be configured incorrectly when LowPower.attachInterruptWakeup() is invoked because the underlying Arduino LowPower_EnableWakeUpPin() function will treat every GPIO pin as a SystemWakeup pin even if it can operate as an ExternalInterrupt pin.  This matters because of this feature of SystemWakeup pins (from the STM32L071 reference manual in this case):

"PWR power control/status register (PWR_CSR)

Bit 8 EWUP1: Enable WKUP pin 1
This bit is set and cleared by software.
0: WKUP pin 1 is used for general purpose I/Os. An event on the WKUP pin 1 does not
wakeup the device from Standby mode.
1: WKUP pin 1 is used for wakeup from Standby mode and forced in input pull down
configuration (rising edge on WKUP pin 1 wakes-up the system from Standby mode).
Note: This bit is reset by a system reset."

So a pin operating in SystemWakeup mode WILL have an internal-pull down attached while the same pin operating in EXTI mode may pull-up, pull-down, or no internal pull-up.  Whether or not this will cause your code to fail to operate will be dependent on the external circuit attached to the pin.  I happened to discover this issue as my particular circuit required the GPIO pin to have the internal pull-up to function properly.  Strictly speaking this isn't a flaw in the STM32LowPower module per se, as it's the underlying Arduino STM32 code that doesn't recognize that GPIO pins have multiple wake-up modes with conflicting properties.  As-is, LowPower.attachInterruptWakeup() should only be called if the deepSleep() function is to be used, which currently puts the STM32 into Standby mode which requires external interrupts to be configured as SystemWakeup.

ABOSTM, 2020-03-26:
Hi @Eqqman ,
Thanks for highlighting this limitation.
I proposed a enhancement to solve this. 
#20
Can you test it, and tell me if it works for you ? 
You need to add a new parameter to attachInterruptWakeup():
ex: `attachInterruptWakeup(  ...... , SLEEP_MODE)` and you should not have the pullup but still the ability to wakeup from all mode except shutdown

closed 2020-03-26 by fpistm

Eqqman, 2020-03-26:
> Hi @Eqqman ,
> Thanks for highlighting this limitation.
> I proposed a enhancement to solve this.
> #20
> Can you test it, and tell me if it works for you ?
> You need to add a new parameter to [...]

Hello-
In my project I set the device to Stop mode and allow waking from either the RTC alarm or EXTI pin.  If I switch from using attachInterrupt(...) to attachInterruptWakeup(..., DEEP_SLEEP_MODE) the processor wakes via the EXTI pin but not the RTC alarm, indicating this is no longer getting set properly.

Eqqman, 2020-03-26:
Update- I did get everything to work with the revised files, perhaps the build wasn't cleaned properly when I switched over.  Based on the limited testing for my specific project the fix appears good.

PR #20 (2020-03-26): attachInterruptWakeup() add parameter to define LowPowerMode
attachInterruptWakeup() add parameter to define LowPowerMode
    
Up to now, when configuring pin for wakeup, if pin has system wakeup capability (able to wakeup from Shutdown), this capabilty is enable (even if we only want to go in idle, sleep or depsleep).
It has the inconvenient to enable automaticvally pull down on the pin, on some STM32 families like L0.

Aim of this patch is to be able to configure pins to wakeup from idle, sleep or deepsleep, but without activating System wakeup capability.
It means it will not be able to wak [... cut]
commit 77e567b1 (2020-03-26): attachInterruptWakeup() add parameter to define LowPowerMode

Up to now, when configuring pin for wakeup,
if pin has system wakeup capability (able to wakeup from Shutdown),
this capabilty is enable (even if we only want to go in idle, sleep or depsleep).
It has the inconvenient to enable automaticvally pull down on the pin,
on some sTM32 families like L0.

Aim of this patch is to be able to configure pins to wakeup from
 idle, sleep or deepsleep, but without activating System wakeup capability.
It means it will not be able to wake from Shutdown.
And thus there is no pull down activated.

For  [... cut]