stm32duino/STM32LowPower #21: How to get elapsed time in low power modes?

wdx04, 2020-04-27:
Hi,
I normally use Arduino's millis() function to measure the elapsed time since application started. But, by entering any of the low-power modes, the millis() counter seems to be freezed, so I can't get the real elapsed time by millis(). 
Is there a way in which I can get the elapsed time even in low-power modes?
Thanks.

fpistm, 2020-04-27:
I did not check but millis() relies on the systick and it is no more incremented when switching to LP mode.
Maybe you can try to use the RTC.

angelnu, 2020-05-24:
I corrected the millis function by adding this code after a deep sleep:

```
    extern __IO uint32_t uwTick;
    uwTick += millis;
```

With this I am able to successfully use Lorawan  with the STM32LowPower library in a similar way as seen at https://www.thethingsnetwork.org/labs/story/a-cheap-stm32-arduino-node.

I can create a PR to add the above code to the LowPower.deepSleep() function. I can also make this fix optional with a control parameter (that I would default to true)

fpistm, 2020-05-26:
you can always create a PR anyway, I don't think it is reliable for all case, what happen if you wake up before the programmed time elapsed ?

angelnu, 2020-05-26:
You are right: this works when no other interrups are configurred beyond the alarm. I use the getEpoch to calculate the elapsed delay I add to uwTick.

I think the key is to calculate the elapsed time accurately: I am around a few 1-5 ms which is good enough for most cases but I have use cases where the precision has to be <1ms.

The minimum delay I can achieve is 15 ms because there is some delays in the HAL wake up.

wdx04, 2020-06-21:
It seems mbed OS's "rtos::Kernel::get_ms_count()" function can get correct elapsed time after deep sleep, by setting up a timer before going into deep sleep.

tcampos123, 2020-07-08:
Want to know how LowPower.shuntdown allows you to wake up by interruption. Because I tested with the blue pill board and it didn't work. I need to save on battery.

daybyter, 2021-03-25:
> Hello Guys,
> 
> Want to know how LowPower.shuntdown allows you to wake up by interruption. Because I tested with the blue pill board and it didn't work. I need to save on battery.

Over here my blue pill wakes up via RTC after shutdown. But I'm somewhat unsure about the current consumption. I currently measure about 39 microAmpere in shutdown mode. According to the datasheet, they talk about 2 microAmp ? And here you see similar results:

https://www.youtube.com/watch?v=TFkU3QD71Pc

So I wonder what your results are? Just to know, if my measuring concept, my board or the software is the actual problem.

I power my board via 3.3V pin, but the 5 => 3.3 V regulator is still on the board. Maybe it is the problem?

daybyter, 2021-03-25:
You use a STM32L0 ? I have a stm32F103 here, which will draw more current.

But I got a recommendation, that I should remove the LDO and it's capacitors if it is not used.

Thanks a lot for your information. Unfortunately your MCU is a bit underpowered for my problem, I guess. I need a camera attached as a sensor. Otherwise I could build on top of it.

closed 2021-09-27 by fpistm