stm32duino/STM32LowPower #27: Nucleo L432KC low power problem

huiskesmartijn, 2020-06-09:
I’m trying to go in sleep mode (using the LowPower.deepSleep() command) for a few seconds between 2 serial prints.

But somehow the print statements are right after eachother, so the board doesn’t appear to go to sleep. Any idea?

fpistm, 2020-06-09:
DeepSleep does not reset, shutdown does.
And as Serial use interrupt it wake up the board directly. Do a Serial.flush() before sleep...

fpistm, 2020-06-10:
For ref:
https://github.com/stm32duino/STM32LowPower/issues/25#issuecomment-633549395

closed 2020-06-10 by fpistm

huiskesmartijn, 2020-06-10:
Hi, thanks for the clarification!

However, even when using the example here led blinks very fast (way more than once per second) https://github.com/stm32duino/STM32LowPower/blob/master/examples/TimedWakeup/TimedWakeup.ino

So it seems my device is not going into deepsleep at al for some reason.

fpistm, 2020-06-10:
Well, I don't know why it wakes up too quickly with your board, probably the same issue, there is an interrupt which fired, I guess. An EXTI ? Did you made some soldier bridge change ? Did you used exactly the TimedWakeup.ino? Which core version? which low power and RTC library version ?

I've tested and it works as expected:
[image]

The fact it is not 1 second is normal as it took some times to wake up.

RobMBSos, 2021-08-05:
@huiskesmartijn Sorry for commenting on this old and already closed issue, I am trying this library in a Nucleo L432KC module but I am not able to make it work. Consumption is around 40mA with peaks of 60mA how did you solve the problem?
Thanks!