stm32duino/STM32LowPower #28: Power consumption for STM32L0 below 1mA

IoTThinks, 2020-06-30:
Hi everyone,
I'm using stm32duino for STM32L0 for Dragino LSN50.
I use the variant B-L072Z-LRWAN1 as it is very similar to LSN50. The only difference are some SPI and LoRa pins.

I'm able to go down from 10mA in normal mode to 1mA in deep sleep / shutdown mode.
LoRa is not in used yet.
Is there anyway I should take note to go down bellow 1mA?
By right, the STM32L) should go to uA in deep sleep/shutdown mode?

Thanks a lot.

```
#include "config.h"
#include "STM32LowPower.h"

// https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/DISCO_L072CZ_LRWAN1
// DISCO Pin: https://os.mbed.com/platforms/ST-Discovery-LRWAN1/
  
void setup() {  
  Serial.begin(BAUD_RATE);
  Serial.println("Hello LoRa Low-Power Node");
  delay(1000);  

  // Configure low power
  LowPower.begin();

  // Setup LoRa
  // setupLoRa();  
}

void loop() { 
  // After sleep, doing some work
  // Serial.println("Doing some work...");
  
  // Send to gateway
  // sendLoRaMessage(sensorMessage);
  // sendLoRaMessage("Hi");
  delay(10000);
 
  // Deep sleep       
  Serial.println("Going to sleep...");  
  // delay(1000);
  // LoRa.sleep();
  // LowPower.sleep(5000);
  // LowPower.deepSleep(10000);
  LowPower.shutdown(10000);
  // LoRa.idle();
}
```

IoTThinks, 2020-06-30:
Need to turn off LoRa before sleeping.

Reach 11.3uA.

closed 2020-06-30 by IoTThinks

fpistm, 2020-06-30:
Hi @IoTThinks 
yes, the library only manage the MCU low power mode, but user have to ensure shutdown all other peripherals and maybe do some other config like put all unused GPIO's in input Analog mode...

IoTThinks, 2020-06-30:
Yeah, thanks a lot.
I dont know if I need to turn off ADC or something more manually?

IoTThinks, 2020-06-30:
Which mode is the deepSleep() and shutdown() equivalent to?
Thanks a lot.

[image]

fpistm, 2020-06-30:
For L0;
Shutdown is standby.
deepSleep is Stop mode.

IoTThinks, 2020-06-30:
Standby and stop mode have 2 modes with/without RTC.
We need to manipulate RTC via STM32RTC, right?
https://github.com/stm32duino/STM32RTC

My board uses 11.3uA. Still far away from 290nA (0.3uA).
Thanks millions.

fpistm, 2020-06-30:
RTC is used for delayed wakeup
As said library manages a default mode for the MCU.
It does not managed all possibilities and could not as it depends of the boards. For example, as described in the HAL:

>       To minimize the consumption In Standby mode, VREFINT, the BOR, PVD, and temperature
>        sensor can be switched off before entering the Standby mode. They can be switched
>        on again by software after exiting the Standby mode.
>        function.

Moreover, I guess you measure the power of the board not the MCU only.

IoTThinks, 2020-06-30:
Yeah, I measure the board not just the MCU.
Let me play around.
11.3.uA is too good for me already.

Thanks a lot for your insight.

evandavey, 2021-07-21:
@IoTThinks - I am using a similar board (with an STM32L0 and SX1262 radio).  I can ony get down to 90uA calling deepSleep.  I know this board is capable of as low as 2uA (this is what is achieved with the supplied firmware).  How did you shut down the LoRa radio?  I am trying to use https://github.com/LacunaSpace/basicmac.

codev123, 2024-02-23:
try to shutdown spi bus