stm32duino/STM32LowPower #34: Bluepill F103C8: Build fails on platformio for version 1.1.0

nbars, 2020-07-15:
Hello,
thank you for providing us with the library. Unfortunately, `STM32duino Low Power@1.1.0` fails to build using the following configuration:

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino

lib_deps =
    STM32duino Low Power@1.0.3
    STM32duino RTC
```

Error message:
```
Compiling .pio/build/bluepill_f103c8/libae0/Wire/utility/twi.c.o
Compiling .pio/build/bluepill_f103c8/lib1bf/STM32duino Low Power/STM32LowPower.cpp.o
Compiling .pio/build/bluepill_f103c8/lib1bf/STM32duino Low Power/low_power.c.o
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::begin()':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:56:3: error: 'LowPower_init' was not declared in this scope; did you mean 'LowPower'?
   56 |   LowPower_init();
      |   ^~~~~~~~~~~~~
      |   LowPower
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::idle(uint32_t)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:71:3: error: 'LowPower_sleep' was not declared in this scope
   71 |   LowPower_sleep(PWR_MAINREGULATOR_ON);
      |   ^~~~~~~~~~~~~~
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::sleep(uint32_t)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:85:3: error: 'LowPower_sleep' was not declared in this scope
   85 |   LowPower_sleep(PWR_LOWPOWERREGULATOR_ON);
      |   ^~~~~~~~~~~~~~
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::deepSleep(uint32_t)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:99:3: error: 'LowPower_stop' was not declared in this scope; did you mean 'LowPower'?
   99 |   LowPower_stop(_serial);
      |   ^~~~~~~~~~~~~
      |   LowPower
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::shutdown(uint32_t)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:113:3: error: 'LowPower_shutdown' was not declared in this scope
  113 |   LowPower_shutdown();
      |   ^~~~~~~~~~~~~~~~~
Archiving .pio/build/bluepill_f103c8/libae0/libWire.a
Archiving .pio/build/bluepill_f103c8/libed8/libSTM32duino RTC.a
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::attachInterruptWakeup(uint32_t, voidFuncPtrVoid, uint32_t, LP_Mode)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:130:5: error: 'LowPower_EnableWakeUpPin' was not declared in this scope; did you mean 'HAL_PWR_EnableWakeUpPin'?
  130 |     LowPower_EnableWakeUpPin(pin, mode);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
      |     HAL_PWR_EnableWakeUpPin
Indexing .pio/build/bluepill_f103c8/libae0/libWire.a
Indexing .pio/build/bluepill_f103c8/libed8/libSTM32duino RTC.a
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp: In member function 'void STM32LowPower::enableWakeupFrom(HardwareSerial*, voidFuncPtrVoid)':
.pio/libdeps/bluepill_f103c8/STM32duino Low Power/src/STM32LowPower.cpp:147:5: error: 'LowPower_EnableWakeUpUart' was not declared in this scope
  147 |     LowPower_EnableWakeUpUart(_serial, callback);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/bluepill_f103c8/lib1bf/STM32duino Low Power/STM32LowPower.cpp.o] Error 1
```

fpistm, 2020-07-16:
Hi,

I've tested with Arduino IDE to build the AlarmTimedWakeup example for the blue pill with:
* Core version 1.9.0.
* LowPower master (will be the 1.1.0)
* RTC 1.0.3

And it build fine.

> "C:\\STM32\\arduino\\arduino-1.8.13\\portable\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-size" -A "C:\\Users\\frq92243\\AppData\\Local\\Temp\\arduino_build_667049/AlarmTimedWakeup.ino.elf"
> Sketch uses 27712 bytes (42%) of program storage space. Maximum is 65536 bytes.
> Global variables use 1252 bytes (6%) of dynamic memory, leaving 19228 bytes for local variables. Maximum is 20480 bytes.
> 

Unfortunately we support only Arduino IDE not PIO.
Anyway I'm surprised that you set the @1.1.0 for LP library as it is not officially released, only available at master.
To manage dependencies each libraries check version so maybe an issue in pio env to find them anyway I'm surprised as this is defining in the include.

nbars, 2020-07-16:
I guess that PIO is using a different version numbering scheme than you are. At least, version 1.1.0 ist the default given on the [example page](https://platformio.org/lib/show/5504/STM32duino%20Low%20Power/examples) of PIO.
I will try to get in touch with the PIO maintainer of `STM32duinoLowPower`, maybe he knows any solution.

fpistm, 2020-07-16:
Ok. In the exemple of installation 1.1.0 is used for illustration. But if you check the version you will see it ends to 1.0.0

KiraVerSace, 2020-10-13:
> I guess that PIO is using a different version numbering scheme than you are. At least, version 1.1.0 ist the default given on the [example [...]

how do you slove it at last?

KiraVerSace, 2020-10-13:
> Ok. In the exemple of installation 1.1.0 is used for illustration. But if you check the version you will see it ends to 1.0.0

See my solution

BitExplorer, 2021-04-06:
Changed my platformio.ini file to reflect the correct version and it works perfectly. 

stm32duino/STM32duino Low Power @ 1.0.3