stm32duino/STM32LowPower #35: Multiple External Interrupt Not working in Deep Sleep Mode

srjtmitra5, 2020-08-14:
In Low Power Deep Sleep mode  in setup multiple external interrupt pins declared. But only one ISR is responding while others are not working. If I pick only one at a time and comment out the rest then it is working. I want to use multiple external interrupts as source of wake-up. Please help!

fpistm, 2020-08-14:
Hi @srjtmitra5 
Could you be more precise about your setup and code ?
Which board? sketch? core version and library?
Thanks in advance.

srjtmitra5, 2020-08-14:
Hello!,
I am using STM32F103RBT6,
Library : https://github.com/stm32duino/STM32LowPower
I am using this example: https://github.com/stm32duino/STM32LowPower/tree/master/examples/ExternalWakeup
It is working great with a single interrupt but failing for multiple.
Thanks for your comment!

srjtmitra5, 2020-08-15:
My apologies. I forgot to mention that in the example we have LowPower.sleep() but I replaced it with LowPower.deepSleep(). Is that a reason?

fpistm, 2020-08-17:
Which pins did you try?

srjtmitra5, 2020-08-19:
I think I have done some mistake in the hardware. I am still checking. For now I am closing this issue unless I am absolutely sure

closed 2020-08-19 by srjtmitra5