stm32duino/STM32LowPower #37: Add support for wake on SPI

llamington, 2020-08-24:
Would it be possible for support to be added for waking on incoming SPI signal?
This library is great by the way!

fpistm, 2020-08-24:
Hi @llamington 
I'm not aware on SPI wake up capabilities.
Anyway you can use on of the wakeup in as CS then you could wake up.

FRASTM, 2022-01-04:
This depends on the low power mode. SPI can exit from LP sleep
When entering STM32 STOP 0/1/2 or standby or shutdown mode the SPI is _not_ able to exit from this low power mode (no wakeup capability unlike the RTC or UART or I2C)
On the other hand, for stop or standby or shutdown modes, some stm32 devices have several Wake Up pins  (example PA0 or PC13 or PE6 or PA2 or PC5) which can exit from stop/standBy/shutdown low power modes.
So I imagine that Routing one of the SPI signal to one of the those wakeUp pin (ex connecting SPI Rx this input) could be tried.

closed 2022-01-04 by fpistm