stm32duino/STM32LowPower #42: Got compilation errors

pankleks, 2020-12-05:
Hello,

I've been trying to use this library with Blue Pill board.

After adding dependency to STM32RTC + Wire, I got following (adding platformio config):

[image]

Thank you in advance for help!

pankleks, 2020-12-05:
when I donwgraded to 1.0.3 all is compiling fine.

but when you install library by platform IO it is installing by default 1.1.0 version.

fpistm, 2020-12-05:
Hi @pankleks 
As explained here:
https://github.com/stm32duino/STM32LowPower/issues/34#issuecomment-707802604

The 1.1.0 is not yet released as it depends on the STM32 core.

closed 2020-12-05 by fpistm