stm32duino/STM32LowPower #45: Bluepill STM32 doesn't enter sleep mode

QuitButton, 2021-01-11:
I'm trying to work out what the default wakeup events are for the STM32.

I tried modifying the timed wakeup example from **deepSleep** to **sleep**, and now it does nothing - sleep mode is never started. 

```
#include "STM32LowPower.h"

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Configure low power
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.sleep(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.sleep(1000);
}
```

FRASTM, 2021-01-12:
@QuitButton , you could please detail a bit, 
 did you change the sleeping time (from 1000ms to x)
 did you find this change caused the system not entering the sleep mode

Manolo8888, 2021-09-26:
_#include "STM32LowPower.h"
int x=1200;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.sleep(x);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.sleep(x);
  delay(200);
}

It doesnt work.
Compiled in Arduino platform

fpistm, 2021-09-27:
@Manolo8888 
I've retest with a BluePill and it works.
I close this issue as it is inactive and it works as expected.

closed 2021-09-27 by fpistm