stm32duino/STM32LowPower #48: Build library error:  error: 'SYS_WKUP1' undeclared (first use in this function)   103 |       case SYS_WKUP1

naigonakoii, 2021-02-06:
When compiling against the newest STM32 core with this library, I get the following error:

 error: 'SYS_WKUP1' undeclared (first use in this function)
  103 |       case SYS_WKUP1 :

Specifically, this command fails (I can repro on the command line alone)
"C:/Program Files/Arm/xpack-arm-none-eabi-gcc-10.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_744812/sketch/build_opt.h" -c -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-ID:\\Igniter\\OneFX" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/avr" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32/LL" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32/usb" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32/OpenAMP" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32/usb/hid" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino/stm32/usb/cdc" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Drivers/STM32F4xx_HAL_Driver/Inc/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Drivers/STM32F4xx_HAL_Driver/Src/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/STM32F4xx/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/OpenAMP" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/OpenAMP/open-amp/lib/include" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/OpenAMP/libmetal/lib/include" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/OpenAMP/virtual_driver" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Middlewares/Third_Party/FatFs/src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\variants\\OneFxBoot/FatFs" -DSTM32F4xx -DARDUINO=10813 -DARDUINO_OneFxFull -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"OneFxFull\"" -DSTM32F405xx -DHAL_UART_MODULE_ENABLED "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\tools\\CMSIS\\5.7.0/CMSIS/Core/Include/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\tools\\CMSIS\\5.7.0/CMSIS/DSP/Include" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\tools\\CMSIS\\5.7.0/CMSIS/DSP/PrivateInclude" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\cores\\arduino" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\variants\\OneFxBoot" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\STM32LowPower\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\STM32RTC\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\OneFx\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\OneFxCommon\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\Wav\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\DataStream\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\NEC\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\GyroAccl\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\Wire\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\SrcWrapper\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\SPI\\src" "-ID:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\DigitalIO\\src" "D:\\My Documents\\Arduino\\hardware\\OneFX\\STM32\\libraries\\STM32LowPower\\src\\low_power.c" -o "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_744812\\libraries\\STM32LowPower\\low_power.c.o"

Where is SYS_WKUP1 actually defined as a variable? I see it gets set in PinDefinitoins.h for each variant, but it never seems to be defined as an extern anywhere.

fpistm, 2021-02-06:
It is defined in the variant in the file PinNamesVar.h.
For example:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/Generic_F4x5RG/PinNamesVar.h

It seems you made your own variant. https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino/STM32F4
You can find a generated file four your mcu here:

naigonakoii, 2021-02-06:
The thing I don't understand is that I havePinNamesVar.h defined in my variant, and SYS_WKUP1 is defined in that file:

#ifdef PWR_WAKEUP_PIN1
    SYS_WKUP1 = PA_0,
#endif
#ifdef PWR_WAKEUP_PIN2
    SYS_WKUP2 = NC,
#endif
#ifdef PWR_WAKEUP_PIN3
    SYS_WKUP3 = NC,
#endif
#ifdef PWR_WAKEUP_PIN4
    SYS_WKUP4 = NC,
#endif
#ifdef PWR_WAKEUP_PIN5
    SYS_WKUP5 = NC,
#endif
#ifdef PWR_WAKEUP_PIN6
    SYS_WKUP6 = NC,
#endif
#ifdef PWR_WAKEUP_PIN7
    SYS_WKUP7 = NC,
#endif
#ifdef PWR_WAKEUP_PIN8
    SYS_WKUP8 = NC,
#endif

However, these are just assignments, and if not set would lead to a runtime error, not the build error that SYS_WKUP1 is not defined from my understanding. So there needs to be an extern for it or something else that actually defines it.

NOTE as well, if I change line 103 in low_power.c to PA_0 which is what the PinNamesVars.h above does, then everything compiles fine.

Also note I did create my variant using that template, but that was about a year ago, and I just now re-sync the STM32 core, which brought on the separate of low_power out of the core which started the issues.

fpistm, 2021-02-06:
PinNamesVar is included here: https://github.com/stm32duino/Arduino_Core_STM32/blob/82cdead6ce9bb3be4af6568626e64ecfaee996cf/cores/arduino/stm32/PinNames.h#L251-L254

You probably have an issue in your variant.

naigonakoii, 2021-02-06:
Redoing my entire variant folder with a new default and copying over all the custom stuff I needed at least makes this build pass. It would be nice in the future if the variants stuff didn't change as drastically, as it is better to not have to redo every custom board that is made.

closed 2021-02-06 by naigonakoii