stm32duino/STM32LowPower #53: STM32 doesn't enter sleep mode when compiling on Windows 10 but does when compiled on Linux

haimiko, 2021-03-16:
Compiling the following deepsleep example, from STM32LowPower, on Windows 10 doesn't work but does on Linux.

For some reason when compiling on Win10, STM32 doesn't go to sleep but does when compiled on Ubuntu 20.04 LTS.  Logical conclusion is something with the toolchain.

```
/*
  TimedWakeup

  This sketch demonstrates the usage of Internal Interrupts to wakeup a chip
  in deep sleep mode.

  In this sketch, the internal RTC will wake up the processor every second.

  This example code is in the public domain.
*/

#include "STM32LowPower.h"

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Configure low power
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.deepSleep(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.deepSleep(1000);
}

```

fpistm, 2021-03-16:
Hi @haimiko 
You don't specify which environment you used, IDE version,  which target(boards),....
anyway I guess you use PIO as you raised also this issue there. Try with Arduino IDE and if you have the issue then we will investigate.
For ref: https://github.com/platformio/platform-ststm32/issues/501#issuecomment-799909748

matthewbatcheler, 2021-04-11:
I am having what I believe to be the same problem, using the same sketch in the Arduino IDE on an STM32f103c8 bluepill. When compiled and uploaded from my win 10 laptop it runs fine (1 sec pulses, as expected). However, on my win 10 desktop, using the same STM32duino lowpower (v. 1.0.3) and STM32duino RTC (v. 1.0.3) versions, the LED _usually_ flickers rapidly - the device is clearly not entering deepSleep. Even more bizzarely, the desktop usually _works as expected_ (i.e. 1 sec flashes) when uploading onto the device when it is working properly - usually when I've uploaded from the laptop. 
This behaviour seems to be the same on PlatformIO, though I've not extensively tested that.
Also note that idle() and sleep() work fine. This seems to happen equally when uploading through dfu and serial. I've tried a bunch of versions of both libraries and haven't got any to work.

haimiko, 2021-04-12:
> 
> 
> Hi @haimiko
> You don't specify which environment you used, IDE version, which target(boards),....
> anyway I guess you use PIO as you raised also this issue there. Try with Arduino IDE and if you have the issue then we will investigate.
> For ref: [platformio/platform-ststm32#501 (comment)](https://github.com/platformio/platform-ststm32/issues/501#issuecomment-799909748)

haimiko, 2021-04-12:
> I am having what I believe to be the same problem, using the same sketch in the Arduino IDE on an STM32f103c8 bluepill. When compiled and uploaded from my win 10 laptop it runs fine (1 sec pulses, [...]

Can you please try compiling it on your desktop with Arduino IDE to see if the issue goes away?

matthewbatcheler, 2021-04-13:
Hi
I had been doing that the majority of the time previously. When I tried it again, the light immediately started flickering (i.e. sleep not working). However, after a reset (and fiddling with the wires), it now works and I can't make it fail! 
I am beginning to suspect power or a dodgy connection may be an issue, and could possibly explain why it works better on a different machine? Currently its powered off my dodgy st-link v2 clone, but when I i get a chance I'll power it off a variable supply and see if i can reproduce the behavior with a reduced voltage.
Do you think explanation seems likely? Note the RTC works reliably when using some of the RTC sketches so I don't think that is affected.

fpistm, 2021-09-27:
I've tested on Windows/ Mac and Linux using STM32core 2.0.0 and latest library version and could not reproduce your issue.
As stated, it is probably an hardware issue, LDO, power, STM32 genuine mcu...
So I close it.

closed 2021-09-27 by fpistm

han-something, 2023-09-13:
Had exact the same problem. With different STM32f103c8 bluepill clones and st-link-v2 programmer (IDE = PlatformIO). When doing the LowPower.deepSleep blink, the Led flickers very fast. It works fine when only using LowPower.sleep() though. Solution: just repower your device after programming. Reset doesn't help. You have to remove the power source (programmer, usb cable, whatever) and repower it. Then it works fine.