stm32duino/STM32LowPower #59: LowPowerSupport for STMG0-series

berrak, 2021-07-29:
Hi,
I tried to compile the example sketch 'AlarmTimedWakeup' but it failed with a 
[STM32LowPowerIssue-5.txt](https://github.com/stm32duino/STM32LowPower/files/6901679/STM32LowPowerIssue-5.txt)
linkage error:

      Linking everything together...
      ...
      collect2: error: ld returned 1 exit status
      Using library STM32duino_Low_Power at version 1.1.0 in folder: /home/bekr/MYCODE-ARDUINO-IDE/libraries/STM32duino_Low_Power 
      Using library STM32duino_RTC at version 1.1.0 in folder: /home/bekr/MYCODE-ARDUINO-IDE/libraries/STM32duino_RTC 
      Using library SrcWrapper at version 1.0.1 in folder: /home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper 
      exit status 1
      Error compiling for board Generic STM32G0 series.

My question is then, does the library lack support for STM32G0-series, or is it something that should work, but obviously does not. I'm using Arduino IDE 1.8.15 on Debian Linux/Bullseye. I have attached the complete log from Arduino IDE for inforamtion.

berrak, 2021-07-29:
I use ST-Link (SWD/SWC) and the target is the STM32G031J6M on a breakout board.

ABOSTM, 2021-08-02:
Hi @berrak, 
Yes STM32G0-series is supposed to work.
But you problem is much simplier, from your log:
```
/home/bekr/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_360271/AlarmTimedWakeup.ino.elf section `.text' will not fit in region `FLASH'
/home/bekr/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 6296 bytes
collect2: error: ld returned 1 exit status
```
The interesting part is (hidden) at the line end:
.....AlarmTimedWakeup.ino.elf section `.text' will not fit in region `FLASH'
..... region `FLASH' overflowed by 6296 bytes

So your code is too big for your FLASH device. STM32G0 is small device, and STM32G031J6 has only 32KB flash.
You need to save at least 6296 bytes.

As a start point you can customize your PeripheralPins, and comment out pins you don't use:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/STM32G0xx/G031J(4-6)M_G041J6M/PeripheralPins.c

closed 2021-08-09 by fpistm

berrak, 2021-08-10:
Thanks for the feedback. Much appreciated.