stm32duino/STM32LowPower #61: no matching function for call....

forfrends, 2021-10-04:
Hey! I am trying to compile the standard **SerialDeepSleep** example from the library and I am getting tons of errors:

```
C:\Users\Игорь\Documents\Arduino\libraries\STM32LowPower-master\examples\SerialDeepSleep\SerialDeepSleep.ino: In function 'void setup()':
SerialDeepSleep:25:50: error: no matching function for call to 'STM32LowPower::enableWakeupFrom(USBSerial*, void (&)())'
In file included from C:\Users\Игорь\Documents\Arduino\libraries\STM32LowPower-master\examples\SerialDeepSleep\SerialDeepSleep.ino:13:
C:\Users\�����\Documents\Arduino\libraries\STM32LowPower-master\src/STM32LowPower.h:94:10: note: candidate: 'void STM32LowPower::enableWakeupFrom(HardwareSerial*, voidFuncPtrVoid)'
   94 |     void enableWakeupFrom(HardwareSerial *serial, voidFuncPtrVoid callback);
      |          ^~~~~~~~~~~~~~~~
C:\Users\�����\Documents\Arduino\libraries\STM32LowPower-master\src/STM32LowPower.h:94:43: note:   no known conversion for argument 1 from 'USBSerial*' to 'HardwareSerial*'
   94 |     void enableWakeupFrom(HardwareSerial *serial, voidFuncPtrVoid callback);
      |                           ~~~~~~~~~~~~~~~~^~~~~~
C:\Users\�����\Documents\Arduino\libraries\STM32LowPower-master\src/STM32LowPower.h:95:10: note: candidate: 'void STM32LowPower::enableWakeupFrom(STM32RTC*, voidFuncPtr, void*)'
   95 |     void enableWakeupFrom(STM32RTC *rtc, voidFuncPtr callback, void *data = NULL);
      |          ^~~~~~~~~~~~~~~~
C:\Users\�����\Documents\Arduino\libraries\STM32LowPower-master\src/STM32LowPower.h:95:37: note:   no known conversion for argument 1 from 'USBSerial*' to 'STM32RTC*'
   95 |     void enableWakeupFrom(STM32RTC *rtc, voidFuncPtr callback, void *data = NULL);
      |                           ~~~~~~~~~~^~~
exit status 1
no matching function for call to 'STM32LowPower::enableWakeupFrom(USBSerial*, void (&)())'

```

What could be the problem?
I am using the core: https://github.com/stm32duino/Arduino_Core_STM32
Microcontroller: STM32L152CBT6

fpistm, 2021-10-04:
Hi, 
The default example used the generic "Serial" instance which by default is a HardwareSerial instance but when you enable the CDC then the 'Serial" instance is mapped to the USBSerial instance as stated in the error.
The library does not support wakeup for USB but form U(S)ART (when MCU has this capabilities).
So use a HardwareSerial instance as parameter of the` enableWakeupFrom()`.

closed 2021-10-04 by fpistm

forfrends, 2021-10-04:
Then I have another question:
How to organize wake-up on USB (UART) connection? If I specify the sleep mode **SLEEP_MODE ( sleep ()**), then when the USB is connected, the microcontroller "wakes up" and does not sleep while the USB is connected and the Port Monitor is open on the computer.
If I specify the sleep mode **DEEP_SLEEP_MODE ( deepSleep () ),** then the microcontroller does not react to the USB connection. The microcontroller goes to sleep even if the Port Monitor is open on the computer.
How to make the microcontroller wake up from USB from DEEP_SLEEP_MODE (_as with SLEEP_MODE_)?
Will this work with RTC and LSE? Since I needed the clock source for the RTC it was LSE.

P.S. this method throws errors:
`LowPower.enableWakeupFrom(&**HardwareSerial**, SerialWakeup);`
```
D:\TEMP\arduino_modified_sketch_100407\SerialDeepSleep.ino: In function 'void setup()':
SerialDeepSleep:25:44: error: expected primary-expression before ',' token
   25 |   LowPower.enableWakeupFrom(&HardwareSerial, SerialWakeup);
      |                                            ^
exit status 1
expected primary-expression before ',' token
```

fpistm, 2021-10-04:
GH issue is not to request support. Use the forum instead.
About your error this is normal as HardwareSerial is a class name not an object.
Wake up from USB is not managed by this library, maybe you can try to use the HAL/LL to achieve this but I never tried this.