stm32duino/STM32LowPower #64: STM32L4xx does not restart or resume code execution after wakeup from deep sleep

ssozonoff, 2021-11-30:
Hi,

I have a STM32L4R5ZIY6 which I am trying to get working correctly with deep sleep mode. I can effectively put the MCU into deep sleep mode as observed by the low current readings and it appears to wake up as well per current readings but no code gets executed. According to your read me a wake-up from deep sleep will trigger a reboot. 

I see neither this nor resuming from after the sleep statement.

Thanks for any pointers.
Serge

ssozonoff, 2021-12-02:
Further testing shows the following 

idle(5000) works as expected
sleep(5000) works as expected

deepSleep(5000) the device seems to wakeup from a current consumption point of view but no code gets executed
shutdown(5000) the devices reboots immediately instead of respecting the timeout, shutdown() produces the same result

Thanks,
Serge

FRASTM, 2021-12-13:
With the stm32L4+ MCU, the low power mode STOP2 differs from STOP0 or STOP1  in the way SRAM3 is powered.
When the mcu enters the LowPower_stop(), then 
keeping the SRAM3 content with `HAL_PWREx_EnableSRAM3ContentRetention` can make the ExternalWakeup.ino  HAL_PWREx_EnableSRAM3ContentRetentionenter/exit low power STOP 2 correclty
```
@@ -295,6 +295,10 @@ void LowPower_stop(serial_t *obj)
       || (WakeUpUart->Instance == (USART_TypeDef *)LPUART2_BASE)
 #endif
      ) {
+#if defined(PWR_CR1_RRSTP)
+    // STM32L4+ mcu must keep SRAM3 conetnt is
+    HAL_PWREx_EnableSRAM3ContentRetention();
+#endif /* PWR_CR1_RRSTP */
     // STM32L4xx supports STOP2 mode which halves consumption
     HAL_PWREx_EnterSTOP2Mode(PWR_STOPENTRY_WFI);
   } else
```

FRASTM, 2021-12-13:
Another approach for the stm32L4+ is to map the DEEP_SLEEP to the stm32 STOP1 low power mode: replaing the HAL_PWREx_EnterSTOP2Mode with HAL_PWREx_EnterSTOP1Mode for this serie.
In terms of current consumption: stop1 or stop2 + SRAM3 powered are similar.

ssozonoff, 2021-12-13:
Thanks I will try that then

> In terms of current consumption: stop1 or stop2 + SRAM3 powered are similar.

Any difference in wake-up time or behaviour ?

Serge

FRASTM, 2021-12-13:
The WakeUp time is not increased between STOP2 w or w/o SRAM3. WakeUp Time in STOP1 or STOP2 are similar.
(refer to the  [datasheet](https://www.st.com/resource/en/datasheet/stm32l4r5vi.pdf) "Low-power mode wakeup timings")
Depending on your application (max ram size), you can choose to optimize the power consumption in  limiting the SRAM to SRAM1+SRAM2. Then STOP2 is the deep_sleep with SRAM3 not powered at all.

ssozonoff, 2021-12-13:
OK thanks for your help. I can confirm that the MCU now seems to wake up correctly with the first change you suggested but I am not seeing anything close to the low power numbers in the datasheet.

2.8 μA Stop 2 with RTC

I am seeing ~300uA in LowPower_stop() running the TimedWakeup code with longer sleep times and no LED. 

I am testing a bare board (Swan from blues io https://blues.io/products/swan/) with nothing but a battery hooked up to it.
The problem with these STM32's is the sleep modes are so complex and it's not really one size fits all.

FRASTM, 2021-12-14:
Optimizing the power consumption is a long way and widely depends on the hardware. Does that value  only reflect the mcu power consumption ?

FRASTM, 2021-12-21:
@ssozonoff can you confirm that the changing the loop of the sketch in that way
```
@@ -46,8 +46,7 @@ void loop() {
   }
   // Triggers an infinite sleep (the device will be woken up only by the registered wakeup sources)
   // The power consumption of the chip will drop consistently
-  //  LowPower.sleep();
-  LowPower.deepSleep(6000);  /* deep sleep during 6 sec and exit on RTC Alarm */
+  LowPower.sleep();
 }
```

will enter/exit deep sleep (STOP2) correclty. 
This is the case on my target stm32l4s5 disco kit

ssozonoff, 2021-12-22:
Hi, yes that works. 

Thanks

FRASTM, 2021-12-22:
Good. 
So this issue will be closed once the PR #66 is merged

PR #66 (2022-01-03): stm32L4+ going to STOP2 lowpower mode must keep SRAM3
On the stm32L4+ , the low power DEEP_SLEEP mode corresponds to STOP2 low power
mode. In this case, the SRAM3 must be powered during the low power phase since it is a part of the SRAM area for this target (SRAM is made of SRAM1+SRAM2+SRAM3).
By default, the content of the SRAM3 is lost during STOP2 low power mode and consequently this must be changed. 

Fixes #64
closed 2022-01-03 by fpistm