stm32duino/STM32LowPower #67: Serial is not output normally when returning from DeepSleep

nopnop2002, 2022-02-27:
Serial is not output normally when returning from DeepSleep

# Envitonment
STM32F103 + platformIO

```
Platform Manager
================
Platform ststm32
--------
Checking platformio/ststm32                   15.2.0                             [Up-to-date]
Checking platformio/toolchain-gccarmnoneeabi  1.70201.0 @ >=1.60301.0,<1.80000.0 [Up-to-date]
Checking platformio/framework-arduinoststm32  4.20100.211028 @ ~4.20100.0        [Up-to-date]
Checking platformio/framework-arduinoststm32-maple 3.10000.201129 @ ~3.10000.0        [Up-to-date]
Checking platformio/tool-stm32duino           1.0.1 @ ~1.0.1                     [Up-to-date]
Checking platformio/tool-openocd              2.1100.211028 @ ~2.1100.0          [Up-to-date]
Checking platformio/tool-dfuutil              1.9.200310 @ ~1.9.190708           [Up-to-date]
```
# My code
``
#include <Arduino.h>
#include "STM32LowPower.h"

#define BAUDRATE 115200

void setup() {
  LowPower.begin();
  Serial.begin(115200);
  Serial.println("serial (same as 1)");
  Serial.println("*****");
}

void loop() {
  Serial.println("serial printing");
  for(int i=0;i<20;i++) LowPower.deepSleep(1000);
  //for(int i=0;i<20;i++) LowPower.sleep(1000);
  //for(int i=0;i<20;i++) LowPower.idle(1000);
  //delay(1000);
}
```

# problem

Serial is not output normally when returning from DeepSleep.

Serial is output normally when returning from Sleep.

Serial is not output normally when return

![LowPower](https://user-images.githubusercontent.com/6020549/155875370-6344b918-abf6-48e6-85a5-4e958f0221fb.jpg)
ing from Idle.

fpistm, 2022-02-28:
Same here than #68. Use Serial.flush(). Entering and exiting sleep mode change the clock that's why you probably seen this.

nopnop2002, 2022-02-28:
Good!!
It's solved.
```
void loop() {
  //unsigned long nowMillis = millis();
  //Serial.println(nowMillis);
  Serial.println("serial printing");
  Serial.flush();
  for(int i=0;i<20;i++) LowPower.deepSleep(1000);
  //for(int i=0;i<10;i++) LowPower.sleep(1000);
  //for(int i=0;i<5;i++) LowPower.idle(1000);
  //delay(1000);
}
```

closed 2022-02-28 by nopnop2002