stm32duino/STM32LowPower #70: The sleep time is incorrect.

nopnop2002, 2022-02-28:
# My Envitonment
STM32F103(8MHz Xtal) + platformIO

```
Platform Manager
================
Platform ststm32
--------
Checking platformio/ststm32                   15.2.0                             [Up-to-date]
Checking platformio/toolchain-gccarmnoneeabi  1.70201.0 @ >=1.60301.0,<1.80000.0 [Up-to-date]
Checking platformio/framework-arduinoststm32  4.20100.211028 @ ~4.20100.0        [Up-to-date]
Checking platformio/framework-arduinoststm32-maple 3.10000.201129 @ ~3.10000.0        [Up-to-date]
Checking platformio/tool-stm32duino           1.0.1 @ ~1.0.1                     [Up-to-date]
Checking platformio/tool-openocd              2.1100.211028 @ ~2.1100.0          [Up-to-date]
Checking platformio/tool-dfuutil              1.9.200310 @ ~1.9.190708           [Up-to-date]
```

# My config
```
[platformio]
default_envs = blackpill_f103c8

[env:blackpill_f103c8]
platform = ststm32
board = blackpill_f103c8
framework = arduino
lib_deps =
  https://github.com/stm32duino/STM32LowPower
  https://github.com/stm32duino/STM32RTC
```

# My code
```
#include <Arduino.h>
#include "STM32LowPower.h"

void setup() {
  LowPower.begin();
  Serial.begin(115200);
  Serial.println("serial (same as 1)");
  Serial.println("*****");
}

void loop() {
  unsigned long nowMillis = millis();
  Serial.println(nowMillis);
  Serial.println("serial printing");
  Serial.flush();
  for(int i=0;i<20;i++) LowPower.deepSleep(1000);
  //for(int i=0;i<10;i++) LowPower.sleep(1000);
  //for(int i=0;i<5;i++) LowPower.idle(1000);
  //delay(1000);
}
```

# Problem

My understanding is that the code below sleeps 1000ms 20 times.

However, it actually sleeps only 1000 milliseconds.

fpistm, 2022-02-28:
As stated the millis is based on systick when you are in sleep mode it is not incremented.

Ex displaying the timestamp:

10:23:20.962 -> 1114
10:23:20.962 -> serial printing
10:24:04.114 -> 1336
10:24:04.114 -> serial printing
10:24:47.248 -> 1558
10:24:47.248 -> serial printing
10:25:30.420 -> 1780
10:25:30.420 -> serial printing
10:26:13.533 -> 2002
10:26:13.533 -> serial printing
10:26:56.661 -> 2224
10:26:56.661 -> serial printing

Ok we have more than 20 * 1000 ms but entering and exiting deep sleep take some times.

nopnop2002, 2022-02-28:
@fpistm 

>Ex displaying the timestamp:

I want your code.

nopnop2002, 2022-02-28:
I want to timestamp feature.

fpistm, 2022-02-28:
> I want to timestamp feature.

Well use the Arduino IDE serial monitor.
Your issue here is that you use millis() to know how many time spent but as said millis is based on systick which is disabled (not increment) during the sleep time.

nopnop2002, 2022-02-28:
>Well use the Arduino IDE serial monitor.

I don't use  Arduino IDE.
STM32F103(8MHz Xtal) + platformIO.

nopnop2002, 2022-02-28:
>Note that we officially doesn't support PIO. Only Arduino IDE.

Oh!!
I would like PIO support.
This is the PIO log.
```
[2022-02-28 22:43:21.833] serial printing
[2022-02-28 22:43:21.897] 13313
[2022-02-28 22:43:21.897] serial printing
[2022-02-28 22:43:21.897] 13555
[2022-02-28 22:43:21.897] serial printing
[2022-02-28 22:43:21.897] 13797
[2022-02-28 22:43:21.897] serial printing
[2022-02-28 22:43:21.897] 14039
```

fpistm, 2022-02-28:
Based on your log, you probably have a wakeup event which appears. See all your wakeup pin of this board. 
Moreover is it a guenine STM32, I know some blackpill have no guenine stm32.
And non we do not support PIO so I close this issue as you change the title. Use PIO GH instead.

closed 2022-02-28 by fpistm

nopnop2002, 2022-02-28:
I have one question.
What is "PIO GH"

fpistm, 2022-02-28:
PlatformIO GitHub:

https://github.com/platformio

for stm32:
https://github.com/platformio/platform-ststm32

But I don't think they could help on this. For me it is linked to the hardware.