stm32duino/STM32LowPower #71: Examples of STM32LowPower for STM32WLE5CCU6

DEDP01, 2022-06-06:
Hello!

I prove the examples (AlarmTimedWakeup, ExternalWakeup and TimeWakeup), the first two examples worked correctly but the last one (TimedWakeup) doesn´t do anything, it only turns on the led and it keeps like that. 

The example works correctly or is incompatible or the program has a actualization?

DEDP01, 2022-06-07:
Can you tell us more about your setup ?

Yes, my setup:

- STM32 MCU based boards version 2.2.0
- STM32LowPower version 1.2.0
- STM32RTC version 1.3.0

I guess you are using variant "Generic WL55CCU", can you confirm ?
No , the variant is Generic WLE5CCUx

Yes, I customized the output, in my case LED_BUILTIN I changed to PB5.

`
#include "STM32LowPower.h"

void setup() {
  // in  my case the pin is PB5.
  pinMode(PB5, OUTPUT);
  LowPower.begin();
}

void loop() {
  digitalWrite(PB5, HIGH); //only the led keeps on.
  LowPower.deepSleep(1000);
  digitalWrite(PB5, LOW);
  LowPower.deepSleep(1000);
}
`

ABOSTM, 2022-06-08:
Can you check whether after sketch is loaded in Flash, unplug/plug of PowerSupply make a difference ?

DEDP01, 2022-06-08:
I already tried and it still doesn't work

ABOSTM, 2022-06-13:
@DEDP01,
I proposed a fix in RTC library: https://github.com/stm32duino/STM32RTC/pull/68
You can test it if you want.

closed 2022-06-24 by fpistm