stm32duino/STM32LowPower #73: stm32l412kb nucleo not waking up from sleep mode.

yadunandan6176, 2022-07-12:
I am using Nucleo stml412kb with the STM32LowPower library. I wanted to wake my MCU from sleep mode for every external interrupt. I used one of the examples that is provided by the library (ExternalWakeup). The code works fine for some time and then the MCU won't wake up from sleep mode for the external interrupt, later if the reset button is pressed then only the MCU wakes up from sleep mode.

I am, not able to understand what is the issue. I tried debugging by adding some flags in the ISR and serial print statements in the loop. But still, I could not debug the issue. 

Here is the code I was using :
```
#include "STM32LowPower.h"

// Blink sequence number
// Declare it volatile since it's incremented inside an interrupt
volatile int repetitions = 1;

// Pin used to trigger a wakeup
#ifndef USER_BTN
#define USER_BTN SYS_WKUP1
#endif

const int pin = USER_BTN;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Set pin as INPUT_PULLUP to avoid spurious wakeup
  pinMode(pin, INPUT_PULLUP);

  // Configure low power
  LowPower.begin();
  // Attach a wakeup interrupt on pin, calling repetitionsIncrease when the device is woken up
  // Last parameter (LowPowerMode) should match with the low power state used: in this example LowPower.sleep()
  LowPower.attachInterruptWakeup(pin, repetitionsIncrease, RISING, SLEEP_MODE);
}

void loop() {
  for (int i = 0; i < repetitions; i++) {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
  }
  // Triggers an infinite sleep (the device will be woken up only by the registered wakeup sources)
  // The power consumption of the chip will drop consistently
  LowPower.sleep();
}

void repetitionsIncrease() {
  // This function will be called once on device wakeup
  // You can do some little operations here (like changing variables which will be used in the loop)
  // Remember to avoid calling delay() and long running functions since this functions executes in interrupt context
  repetitions ++;
}
```
Any help would be grateful.

ABOSTM, 2022-07-29:
Hi @yadunandan6176 ,
Can you please provide additional information: 
Which Arduino_Core_STM32 version do you use ?
Which version of the library STM32LowPower ?
`The code works fine for some time`  What is the order of magnitude ? Is it minutes, hours, days ?

ABOSTM, 2022-08-05:
Hi @yadunandan6176,
I reproduced the issue with Core version 2.2.0 and STM32LowPower 1.2.0
Thanks for highlighting this issue.

After deep analysis, I found 2 issues related to implementation of LowPower Sleep mode on some STM32 series, including L4.
Fix is available #74,
feel free to test it.

tshcherban, 2022-08-18:
The same issue for STM32L051C8T6. Tried _TimedWakeup.ino_ example, replaced _digitalWrite_ with _Serial.print()_

ABOSTM, 2022-08-23:
@tshcherban ,
I don't have STM32L051C8T6 but I tested both Nucleo-L010RB and Nucleo-L031K6
Both are working fine with the sketch ExternalWakeup (the one which is concerned by this issue), with the patch https://github.com/stm32duino/STM32LowPower/pull/74,

I also tested TimedWakeup sketch on my Nucleo-L010RB (but compiled with generic variant L051C8Tx),
example works fine, I just needed to add `#define LED_BUILTIN PA5`
No need for extra RTC configuration.
If you still have issue with TimedWakeup example please raise a separate issue.

Concerning shutdown, It seems working properly on my nucleo l010RB with the sketch below.
LED is flashing every 2seconds:
MCU goes in standby just after led ON. This immediately disables the LED (most GPIO are in High impedance during standby), this causes LED flashing
and after 2seconds MCU reboots, and LED flashes again.

```
#define LED_BUILTIN PA5
#include "STM32LowPower.h"

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Configure low power
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.shutdown(2000);
}
```

Try this sketch (with the patch https://github.com/stm32duino/STM32LowPower/pull/74,) and if it is still not working, please raise a new issue.

closed 2022-08-23 by fpistm