stm32duino/STM32LowPower #79: TimedWakeup not working

tshcherban, 2022-09-02:
I've tried a very basic example of timed wakeup:
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  LowPower.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.sleep(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.sleep(1000);
}
```
And its working wrong. .Works as expected on first board power-on, but on subsequent resets via rst button - LED stays on forewer. Also tried `deepSleep`, `shutdown', `idle` - behavior is the same.
And if i put that before `LowPower.begin();` call:
```
STM32RTC &rtc = STM32RTC::getInstance();
rtc.begin(true); // note parameter here - tells RTC library to do time reset
```
all works as expected both on first power-on and subsequent resets. (of course shutdown reboots board so LED blinks shortly instead of cycling 1 sec, that im aware of). Im not sure, maybe there is some bug in RTC library? or LowPower should care about some additional stuff?
Verified on:
STM32L051C8T6 custom board, basically bare chip with minimum required R/C
STM32L053R8 nucleo-64
STM32F446RE nucleo-64

tshcherban, 2022-09-04:
**Update:**
_TimedWakeup_ example works good with Stm32RTC v.1.2.0 (internally marked as 1.1.1 for mistake) from [releases page](https://github.com/stm32duino/STM32RTC/releases). Will try finding exact commit which brakes functionality.

ABOSTM, 2022-09-07:
Hi @tshcherban,
I tested example TimedWakeup  (deepSleep) on my Nucleo STM32F446RE, and it is working fine, even after reset button is pressed.

Please note that I used latest commit from both libraries:
* STM32LowPower sha1:1a17d4141b3b9def5cc9bd0b8dd5fc9663a7f07a
Which includes several recent fixes:  
#74 
#77 
#78
* STM32RTC sha1: f620b534f7bffe24f031fc5935324027cfe51320

Can you please test the same ?

tshcherban, 2022-09-10:
Okay, tested (using latest changes).
That example is ok:
```
digitalWrite(LED_BUILTIN, HIGH);
LowPower.deepSleep(1000);
digitalWrite(LED_BUILTIN, LOW);
LowPower.deepSleep(1000);
```
but that - not:
```
digitalWrite(LED_BUILTIN, HIGH);
LowPower.deepSleep(1000);
digitalWrite(LED_BUILTIN, LOW);
delay(1000);
LowPower.shutdown(1000);
```
LED should be 1s ON 2s OFF, but instead it is 1s ON 1s OFF, so `LowPower.shutdown` doesnt respect passed time value and reboots CPU immediately.

ABOSTM, 2022-09-14:
Hi @tshcherban ,
You are right, when `LowPower.shutdown()` is invoked after `LowPower.deepSleep()`,
MCU wakes up immediately.
This was due to WakeUp flag that remained set.

I pushed a fix to clear this flag, and it is now working fine on my side:
https://github.com/stm32duino/STM32LowPower/pull/81
Feel free to test it on your side.

closed 2022-09-27 by fpistm

tshcherban, 2022-09-28:
Yep, works great. Verified on both `STM32F446RE` and `STM32L053R8`. Maybe its time to release a new version of the lib? There's a few bugfixes after `1.2.0`

PR #81 (2022-09-27): fix: clear Wakeup Flag before entering Standby or Shutdown mode
fix: clear Wakeup Flag before entering Standby or Shutdown mode

Fixes #79
commit e11b7d12 (2022-09-27): fix: clear Wakeup Flag before entering Standby or Shutdown mode

Fixes #79