stm32duino/STM32LowPower #80: Does not compile with latest STM32RTC library

hobbya, 2022-09-12:
STM32LowPower: v1.2.0
STM32 Core v1.9.0
Arduino IDE: v1.8.19
Device 1: STM32F103C8 Blue Pill
Device 2: STM32F411CE Black Pill
Code: ExternalWakeUp example

LowPower library will not compile on STM32F103C8 for RTC library v1.2.0 and onward. No problem up to v1.1.1.
[STM32F103.txt](https://github.com/stm32duino/STM32LowPower/files/9545210/STM32F103.txt)
[STM32F411.txt](https://github.com/stm32duino/STM32LowPower/files/9545211/STM32F411.txt)

LowPower library will not compile on STM32F411CE for RTC library v1.3.0 and onward. No problem up to v1.2.0.

fpistm, 2022-09-12:
Hi. Pleasz use the latest core version.

fpistm, 2022-09-12:
As the BSP has been moved to each libraries to avoid core dependencies. Using the core 1.9.0 used the old BSP and so the new API available in RTC library are not available. That's why you got thoses errors.

closed 2022-09-14 by fpistm