stm32duino/STM32LowPower #85: Serial port

Lizethgm, 2023-01-06:
I have been trying to test the low power along with the serial ports.
I am using the Generic STM32WL series board "Generic WLE5CCUx" 
My program is this:

#include "STM32LowPower.h"

String command = "Hello";

void setup() {
  Serial.begin(2400);
  Serial1.begin(2400);
  LowPower.begin();
}

void loop() {
  readMeter();
  delay(500);
  LowPower.deepSleep(10 * 1000);
}

void readMeter()
{  
  Serial1.print(command);
  while(Serial1.available()> 0) 
  {
    Serial.println(Serial1.read());
  }
}

I was trying to test it with this code but it only sends the "Hello" command the second time and that's it. If I remove the low power it works correctly. 
How can I make the command "Hello" that I set continue to be sent?

fpistm, 2023-01-09:
Hi @Lizethgm 
Where `Serial1` is defined or how?

fpistm, 2023-01-09:
Hi @Lizethgm 
I've made a fix in #87.
I've tested on a WB. I will test WL tomorrow.

PR #87 (2023-01-10): fix: set CPU2 low power mode for STM32WBxx
Fixes #85

Typically, action performed by CPU2 on a dual core application. Since this example is single core, perform it by CPU1.

On STM32WB, both CPU1 and CPU2 must be in low-power mode to set the entire System in low-power mode, corresponding to the deepest low-power mode possible.
For example, CPU1 in Stop2 mode and CPU2 in Shutdown mode will make system enter in Stop2 mode.
commit ee9a8e63 (2023-01-10): fix(WB): set CPU2 low power mode

Fixes #85

Typically, action performed by CPU2 on a dual core application.
Since this example is single core, perform it by CPU1.

On STM32WB, both CPU1 and CPU2 must be in low-power mode
to set the entire System in low-power mode, corresponding to
the deepest low-power mode possible.
For example, CPU1 in Stop2 mode and CPU2 in Shutdown mode
will make system enter in Stop2 mode.
closed 2023-01-10 by fpistm