stm32duino/STM32LowPower #89: deepSleep takes effect after power down

xvedra, 2023-03-20:
Hi,
I am working with deepSleep in a BluePill and I have notice that it takes effect only after a power down. Procedure:
1) BluePill power on from St-Link.
2) Clear flash memory
3) Program deepSleep example and does not take effect.
4) Power down: Switch off and switch on St-link . After that it takes effect.
Thank you

fpistm, 2023-03-20:
Hi @xvedra 
Howyou flash it?

xvedra, 2023-03-20:
Hi @fpistm,
IDE 1.8.19 Under Windows 10 pro
upload method: SWD (ST-LINK)
Target: BluePill F103C8
What happen if you manually reset it (not power down)? The same...

After read this:
https://github.com/stm32duino/STM32LowPower/issues/36

Works fine adding HAL_SuspendTick();  before LowPower.deepSleep(); 

Thank you!

fpistm, 2023-03-20:
You used LSE?

xvedra, 2023-03-20:
BluePill board has LSE but I have used the simplest example without LSE:

void setup()
 {
pinMode(LED_BUILTIN, OUTPUT);
LowPower.begin();
}

void loop() 
{
  digitalWrite(LED_BUILTIN, HIGH);
  HAL_SuspendTick(); //added
  LowPower.deepSleep(2000);
  HAL_ResumeTick(); //added
  //LowPower.sleep();
  digitalWrite(LED_BUILTIN, LOW);
  HAL_SuspendTick();  //added
  LowPower.deepSleep(2000);
  HAL_ResumeTick(); //added
  //LowPower.sleep();
}

There was no problem using with LowPower.sleep()

fpistm, 2023-03-20:
Is it a legacy chip? Not a clone?

fpistm, 2023-03-21:
I've tested on my side and have no issue.
Ensure to have latest library version, latest core version and  latest STM32CubeProgrammer as you used SWD.
I've tested with different setup, USB enabled or not,... all works.
Moreover it seems strange using `HAL_SuspendTick` allows to work as the `deepSleep `disable the irq before entering in low power.

xvedra, 2023-03-21:
Is it a legacy chip? Not a clone?
_My BluePill is from Aliexpress. I don't know if it's a clone. I have a custom board with a legacy STM32F108CB with the same result._
_Ensure to have latest library version, latest core version and latest STM32CubeProgrammer as you used SWD..
Latest versions_ 
Moreover it seems strange using HAL_SuspendTick allows to work as the deepSleep disable the irq before entering in low power.
Yes, but only in

fpistm, 2023-03-21:
I've made the same test than you and it works.
I've tested with several BluePill, MapleMini and STM32F103ZE and all works. 
So I have no more clue except if it is a clone and in that case could explain why it does not work. You could try to ask on the forum maybe someone already met this issue.
I close it as it works.

closed 2023-03-21 by fpistm