stm32duino/STM32LowPower #90: Memory consumption

tshcherban, 2023-03-26:
Hi
I have a project using this library (btw thanks for your recent bugfixes, now sleep modes are acting as needed) and lots of other hardware (LoRa, sensors).
At some point stuck because i was running out of flash memory. Quite fast I've discovered that the largest consumption is done by two functions: `mktime `(6572 bytes) and `gmtime` (9868 bytes), which in turn are used for time calculation from human representation to epoch one and vice-versa.
Here is the steps which are executed to set a wakeup alarm:
1. get rtc actual time (in <DATE>-HH:mm:ss plus subseconds)
2. convert it with `mktime` to the epoch representation
3. add number of milliseconds to that time
4. set epoch alarm which in turn uses `gmtime` to convert back to <DATE>-HH:mm:ss

Seems very wasteful to spend ~16kb of flash to just do some time math.
Maybe we can add it by hand?
i.e.
```
  if (ms != 0)
  {
      uint16_t subSeconds = ms % 1000;

      ms = ms / 1000;
      uint8_t day = ms / 86400;
      uint8_t hours = (ms - day * 86400) / 3600;
      uint8_t minutes = (ms - day * 86400 - hours * 3600) / 60;
      uint8_t seconds = (ms - day * 86400 - hours * 3600 - minutes * 60);

      rtc.syncDate();
      rtc.syncTime();

      uint32_t ss = rtc.getSubSeconds() + subSeconds;
      if (ss >= 1000)
      {
        ss -= 1000;
        seconds++;
      }

      uint8_t s = rtc.getSeconds() + seconds;
      if (s >= 60)
      {
        s -= 60;
        minutes++;
      }

      uint8_t m = rtc.getMinutes() + minutes;
      if (m >= 60)
      {
        m -= 60;
        hours++;
      }

      uint8_t h = rtc.getHours() + hours;
      if (h >= 24)
      {
        h -= 24;
        day++;
      }

      rtc.setAlarmTime(h, m, s, ss);
      rtc.setAlarmDay(day);
      rtc.enableAlarm(STM32RTC::Alarm_Match::MATCH_DHHMMSS);
  }
```
That is a code block I have actually used for experiments and seems to be working. It replaces a logic [here](https://github.com/stm32duino/STM32LowPower/blob/ef798e3200a8d136f0062eaeeeac3076433ac0e3/src/STM32LowPower.cpp#L210).
Of course its just a draft, need to add month/year handling (if its not ignored by STMs RTC, not sure?). Also I would rather add extended RTC library API to set all those parameters at once and avoid explicit or multiple internal calls to RTC_SetXXX and SyncTime/Date.

tshcherban, 2023-03-27:
I mean maybe we can cut out that 'known topic'? Or you would prefer to stay on the safe side with epoch approach?

fpistm, 2023-03-27:
Main concern is about performance, I think using date and time would take more time to execute but is it really an issue? 😉 
Do not hesitate to provide a PR then we will review and test.

Setting RTC time to `00:00:00` is correct only if it is not already initialized.
Yes the wake up from RTC clocked by external oscillator is supported, some mode even required specifically to use LSE:
https://github.com/stm32duino/STM32LowPower/blob/ef798e3200a8d136f0062eaeeeac3076433ac0e3/src/STM32LowPower.cpp#L187-L207

tshcherban, 2023-03-27:
Ok, I'll create a PR and also measure a performance. BTW `mktime` and `gmtime` are taking ~124 and ~60 microseconds respectively (measured using a `micros()` call), so in total its like ~180 us. Will compare it to a custom approach.

tshcherban, 2023-03-28:
Preliminary results are: 173 us for epoch approach vs 64 us for custom calculation approach (measured the part with epoch/custom calculation, custom calculation includes `getTime`/`getDate` calls). Also reduced by ~15k of flash and 350 bytes of RAM.
All above valid for STM32L051C8/STM32L053C8, optimize -Os, Newlib Nano.