stm32duino/STM32LowPower #98: STM32F103 Consumption arround 300uA

neryvini, 2023-07-18:
I am trying to use the STOP mode with a STM32F103CBT6, but even after set all gpio set as input analog i am getting 300uA consumption. It should be arround 30uA (with standby mode i got 10uA, testing in the same way).

I cant use standby because it must wake up from PB1, not PA0 and we should know if the wakeup source was an interrupt in PB1 or a RTC Alarm. With standby we can't know it.

Going further i tried to disable peripherals, but i do not know if i did it in the proper way. 
For example how can i disable Serial1 and i2c after init both? In programming manual they say to only set gpio as input_analog and disable dac.

`#include "STM32LowPower.h"
#include <STM32RTC.h>

STM32RTC& rtc = STM32RTC::getInstance();

static uint32_t atime = 1;
static byte seconds = 0;
static byte minutes = 0;
static byte hours = 0;

static byte weekDay = 1;
static byte day = 1;
static byte month = 1;
static byte year = 18;

void setGPIOModeToAllPins();

void RTChndl(void *data) {
}

void PINhndl() {
}

void setup() {
rtc.begin();
rtc.setTime(hours, minutes, seconds);
rtc.setDate(weekDay, day, month, year);

LowPower.begin();
LowPower.attachInterruptWakeup(PB1, PINhndl, FALLING);
LowPower.enableWakeupFrom(&rtc, RTChndl, &atime);
}

void loop() {
HAL_ResumeTick(); //----- Allready WakedUp
rtc.setAlarmEpoch( rtc.getEpoch() + 60 );
setGPIOModeToAllPins();
HAL_SuspendTick();      //----- Suspend HAL tickets
LowPower.deepSleep();
}

void setGPIOModeToAllPins()
{
    pinMode(PA0, INPUT_ANALOG);
    pinMode(PA1, INPUT_ANALOG);
    //------- SMW-SX1262 ------
    pinMode(PA2, INPUT_ANALOG);
    pinMode(PA3, INPUT_ANALOG);
    //-------------------------
    pinMode(PA4, INPUT_ANALOG);
    pinMode(PA5, INPUT_ANALOG);
    pinMode(PA6, INPUT_ANALOG);
    pinMode(PA7, INPUT_ANALOG);
    pinMode(PA8, INPUT_ANALOG);
    pinMode(PA9, INPUT_ANALOG);
    pinMode(PA10, INPUT_ANALOG);
    pinMode(PA11, INPUT_ANALOG);
    pinMode(PA12, INPUT_ANALOG);
    pinMode(PA13, INPUT_ANALOG);
    pinMode(PA14, INPUT_ANALOG);
    pinMode(PA15, INPUT_ANALOG);
    pinMode(PB0, INPUT_ANALOG);
    pinMode(PB1, INPUT_ANALOG);
    pinMode(PB2, INPUT_ANALOG);
    pinMode(PB3, INPUT_ANALOG);
    pinMode(PB4, INPUT_ANALOG);
    pinMode(PB5, INPUT_ANALOG);
    //---------- I2C ----------
    pinMode(PB6, INPUT_ANALOG);
    pinMode(PB7, INPUT_ANALOG);
    //-------------------------
    pinMode(PB8, INPUT_ANALOG);
    pinMode(PB9, INPUT_ANALOG);
    pinMode(PB10, INPUT_ANALOG);
    pinMode(PB11, INPUT_ANALOG);
    pinMode(PB12, INPUT_ANALOG);
    pinMode(PB13, INPUT_ANALOG);
    pinMode(PB14, INPUT_ANALOG);
    pinMode(PB15, INPUT_ANALOG);
}
`

fpistm, 2023-07-18:
Which board? Core version? Rtc and power library version? Ide?

neryvini, 2023-07-18:
It is a custom board, but with all components removed except crystal and capacitors. I am using this core with STM32LowPower library.

This is my platformio.ini

[env:bluepill_f103c8_128k]
platform = ststm32
framework = arduino
board = bluepill_f103c8_128k
upload_protocol = stlink

fpistm, 2023-07-19:
Use .end function.

neryvini, 2023-08-09:
One of the libraries was using Serial as debug function. I add the Serial.end() and after that the consumption in stop mode is arround 15uA.