stm32duino/STM32LowPower #104: Support for LPTIM

HalfSweet, 2023-11-08:
For some mcu's they come with LPTIM peripherals which are more flexible and lower power consumption than RTC for low power applications. will STM32LowPower library support LPTIM peripherals.

Maybe I can open a new pr, but this may require a new STM32RTC like dependency library, is there any coding specification required in this regard

fpistm, 2023-11-08:
Hi @HalfSweet 
any contribution are welcome.
I guess the first step would be to add the LPTIM in the PeripheralPins.
I don't catch your point about a "new STM32RTC like dependency library"

fpistm, 2024-07-01:
I close this one as not updated since a while and LPTIM not supported by the core.

closed 2024-07-01 by fpistm