stm32duino/STM32LowPower #105: It's something wrong when I use shutdown mode with STM32G030C8T6

sunnyguhz, 2023-12-03:
I take a lowpower test with STM32G030C8T6, idle/sleep/deepsleep is all right. It consumes 7uA of electric current when I use deepsleep mode.
But It consumes 1520-2200uA of electric current when I use shutdown mode, I think smomething wrong with it.

**Desktop (please complete the following information):**
 - OS: [Windows10 22h2]
 - Arduino IDE version: [2.2.1]
 - STM32 core version: [2.6.0]
 - Library version: [1.2.4]
 - Tools menu settings if not the default: [No Serial]
 - Upload method: [SWD]
Optimize:"Smallest (-OS) with LTO"
Board part number:"Generic G030C8Tx"

#include "STM32LowPower.h"

```
void setup() {
  LowPower.begin();
}

void loop() {
  //LowPower.idle(10000);
  //LowPower.sleep(10000);
  //LowPower.deepSleep(10000);
  LowPower.shutdown(10000);
}
```

fpistm, 2023-12-03:
Try without LTO.

fpistm, 2023-12-08:
Hi @sunnyguhz 
I've tested and shutdown works.
Some point to pay attention:
- When RTC is used to wakeup from this mode then the RTC clock source is LSEas all other clock are switched off. If your board do not have LSE then program will simply fail because LSE cannot be started.
- If RTC is not used then no worry, you can use wakeup pin to wkae up from this mode.
[image]

I've made a PR to enhance this mode but there is no issue.

closed 2023-12-08 by fpistm