stm32duino/STM32LowPower #107: SerialDeepSleep not working with user defined Harware Serial ?

Yannick-Marietti, 2024-02-17:
Hello
It seems example SerialDeepSleep.ino does not work with an hardware serial diffrent form the "Serial" default.
For example, on my computer, the attached (picture) code does not run beyond row 33.

[image]

Best regards

Yannick

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Arduino IDE version: 2.3.1
 - STM32 core version: 2.7.1
 - Library version: STM32duino Low Power 1.2.5
 - Upload method: Mass storage

**Hardware (please complete the following information):**
 - Board Name: NUCLEO L476RG
 

**Additional context**
Add any other context about the problem here.

fpistm, 2024-02-17:
The uart have to be low power compliant. Ex lpuart.

fpistm, 2024-02-17:
Extract from L476 ref man
[image]

fpistm, 2024-02-17:
Will chech deeply next week

fpistm, 2024-02-19:
Hi @Yannick-Marietti 
I've tested and it works as expected using UART4 defined manually.
[image]

Just for information about deepsleep and wakeup from serial feature.
If the Serial is a LPUART then mcu will go in stop 2 mode, if not the stop mode 1 is used to be able to wake up.
See datasheet:
[image]
With default Serial of the Nucleo L476RG, Serial is USART2 so STOP1 mode is used same if we use UART4.

closed 2024-02-19 by fpistm