stm32duino/STM32LowPower #110: No serial output after first wake from deepsleep

miroslavpetrov, 2024-07-01:
**Describe the bug**
Using the unmodified example AlarmTimedWakeup.ino from the library and Seeed Lora-E5 Mini board, after the first wake up from deepsleep there is no output on the serial console.

**To Reproduce**
Upload AlarmTimedWakeup example to the board.

**Expected behavior**
I expect the serial port to continue working after wake up

**Desktop (please complete the following information):**
 - OS: Mac OS Sonoma 14.5
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - Library version: 1.2.5
 - Tools menu settings if not the default: Core logs enabled, Optimize: smallest, Runtime Newlib nano, UART: Enabled(generic serial), Board part number - Lora-E5 mini

 - Upload method: manual export compiled binary and drag and drop to STLink drive

**Hardware (please complete the following information):**
 - Board Name: Seeed Wio-E5 Mini based on STM32wle5jc
 - Hardware Revision: v1.0

miroslavpetrov, 2024-07-03:
@fpistm any ideas?

fpistm, 2024-07-03:
No. I will check when I will have some time.

fpistm, 2024-08-14:
Hi @miroslavpetrov 
I've reproduced and try to debug deeply but hard with this series as debug with low power mode is a nightmare (several issue document in the errata sheet).
In the case of STM32WL, the deepsleep rely on STOP 2 mode and so it requires to properly init the Serial after the deepsleep.

PR #112 (2024-08-20): fix(example): properly end Serial when STOP2 mode is used
Fixes #110.
closed 2024-08-20 by fpistm