stm32duino/STM32LowPower #118: wakeup from serial not working on Lora-E5-Mini

naveen-79, 2025-01-21:
The interrupt wakeup using Serial not working on Lora-E5-Mini module.
Tried using the [Serial Deep Sleep Example](https://github.com/stm32duino/STM32LowPower/blob/main/examples/SerialDeepSleep/SerialDeepSleep.ino)
Device goes into deepsleep but does not wake up on serial recieve. Tried the normal Serial.read() script to verify the functionality of uart, that's working fine.

fpistm, 2025-01-21:
Hi @naveen-79 
I've tested the SerialDeep Sleep on a Nucleo WL55JC and it works as expect.
Issue with the Lora E5 Mini is the default Serial instance is not a LPUART then it is normal it does not work.
The Nucleo WL55JC used PA3/PA2 (LPUART1) as default Serial instance.
The LoRa E5 mini used PB7/PB6 (USART1)  as default Serial instance but it is not compatible with serial wakeup.

If you want be able to wakeup from Serial you have to redefine the default Serial instance or defined new one using:
PA2 or PC1 for Tx and PA3 or PC0 for TX.

closed 2025-01-21 by fpistm

naveen-79, 2025-01-21:
updated the serial definition to `HardwareSerial SleepSerial(PA3, PA2);`. still wakeup does not work

fpistm, 2025-01-21:
And you connect your terminal to those pins?
Other possible issue is the lpuart1 clock config. I guess it is not configured to be able to use 9600 baud. You have to ensure the input clock is correct for this baudrate.

naveen-79, 2025-01-21:
how do i check or change the lpuart clock config

fpistm, 2025-01-21:
You can use cubemx and system core clock config is in the variants folder.

fpistm, 2025-01-21:
Will review it tomorrow.  Nevertheless it is not a bug. End user have to ensure to use correct peripheral for his use case and properly configure it. Like it is not possible to provide all configuration by default nor answer each request to configure it. 😉

fpistm, 2025-01-22:
Looking at the clock the freq should be OK.
You can try do this before the `SleepSerial.begin()` :
```
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {};
  /** Initializes the peripherals clocks
  */
    PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_LPUART1;
    PeriphClkInitStruct.Lpuart1ClockSelection = RCC_LPUART1CLKSOURCE_PCLK1;
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
      Error_Handler();
    }
```

naveen-79, 2025-01-22:
in `deepSleep` the device does not wake up, but in `sleep` mode it wakes up.

fpistm, 2025-01-22:
Hard to help now, it works with the Nucleo WL55JC which is very closed.
So except if you made a mistake on your side hard to help.

naveen-79, 2025-01-22:
got it fixed, restarted arduino ide

naveen-79, 2025-01-22:
working in arduino ide (after restart) 😂, not in platformio. I would greatly appreciate any fixes, or troubleshooting steps to address this problem.

fpistm, 2025-01-22:
We support only Arduino IDE.
If it does not work for pio open an issue on their GitHub.
I guess, they do not build excatly the same way or it is not up to date.