stm32duino/STM32LowPower #122: SerialDeepSleep not working on the STM32WLE5JC (LORA E5)

starrobin34, 2025-05-06:
**Describe the bug**
I have tried the example code on the LORA E5 to have it wake up on UART but it's just staying in deep sleep and is not waking up from stop2. Any advise? 

**To Reproduce**
/*
  SerialDeepSleep

  This sketch demonstrates the usage of Serial Interrupts to wakeup a chip
  in deep sleep mode.

  This sketch is compatible only with board supporting uart peripheral in
  stop mode.

  This example code is in the public domain.
*/

#include "STM32LowPower.h"

// Declare it volatile since it's incremented inside an interrupt
volatile int wakeup_counter = 0;

void setup() {
  Serial.begin(9600);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  // Configure low power
  LowPower.begin();
  // Enable UART in Low Power mode wakeup source
  LowPower.enableWakeupFrom(&Serial, SerialWakeup);
  Serial.println("Start deep sleep wakeup from Serial");
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  // Triggers an infinite deep sleep
  // (the device will be woken up only by the registered wakeup sources)
  // The power consumption of the chip will drop consistently
  LowPower.deepSleep();

  Serial.print(wakeup_counter);
  Serial.println(" wake up");

  // Empty Serial Rx
  while(Serial.available()) {
    char c = Serial.read();
    Serial.print(c);
  }
  Serial.println();
}

void SerialWakeup() {
  // This function will be called once on device wakeup
  // You can do some little operations here (like changing variables
  // which will be used in the loop)
  // Remember to avoid calling delay() and long running functions
  // since this functions executes in interrupt context
  wakeup_counter++;
}

Steps to reproduce the behavior:

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 2.3.6]
 - STM32 core version: [e.g. 2.10.1]
 - Library version: [e.g. 1.4.0]
 - Upload method: [e.g. SWD ST-Link]

**Hardware (please complete the following information):**
 - Board Name: [e.g. Makerverse LORA E5]

fpistm, 2025-05-06:
Hi @starrobin34 

Could you be more precise: Which target you used in the Arduino IDE? Options? What is the Serial pins used?....

fpistm, 2025-05-06:
Main issue here is that you use a board based on the same mcu but maybe not compatible.
First I would recommend to check the clock config for the LPUART1 mainly at 9600.

fpistm, 2025-05-19:
Hi @starrobin34 
any update on this?

closed 2025-05-20 by fpistm

starrobin34, 2025-05-20:
Hey, 
yes it is working now but it is a workaround which I am not proud of and I am not entirely sure why it's working but setting it up like this works for a wake on UART and it catches the whole message. 

```cpp
#include "STM32LowPower.h"

volatile int wakeup_counter = 0;
#define LED_PIN PB5

void SerialWakeup() {
  wakeup_counter++;
}

void setup() {

  Serial.begin(115200);  // Use predefined Serial1 (LPUART1 = PB6/PB7)
  delay(2000); 
  LowPower.begin();
  LowPower.enableWakeupFrom(&Serial, SerialWakeup);
  LowPower.attachInterruptWakeup(PB7, SerialWakeup, RISING, DEEP_SLEEP_MODE);

  Serial.println("Start deep sleep wakeup from Serial");
}

void loop() {

  delay(5); 
  Serial.begin(115200);
  LowPower.deepSleep();  // Sleep until UART RX start bit
  //sleeping here...
  Serial.begin(115200); 
  String msg = Serial.readStringUntil('\r'); //'\r'  or '\n'
  Serial.println(msg);

  //   // Stay awake for 2 seconds to collect serial input
  // while (millis() - start < 2000) {
  //     char c = Serial.read();
  //     Serial.print(c); 
  // }

}
```

fpistm, 2025-05-21:
Your code seems strange as you tried to wakeup form PB7 but it is not a wakeup pin.
Did you try to preoperly configure the LPUART clock then instantiate it.

fpistm, 2025-05-21:
I've tested the SerialDeepSleep example with a LORA E5 mini and it works asexpected.
I've just configured the LPUART1 as stated above and change the default Serial to be on PA2/PA3 pins like this:

```c++
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {};
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_LPUART1;
  PeriphClkInitStruct.Lpuart1ClockSelection = RCC_LPUART1CLKSOURCE_LSE;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
      Error_Handler();
  }
  Serial.setTx(PA2);
  Serial.setRx(PA3);
  Serial.begin(9600);
```

> 14:03:16.820 -> Start deep sleep wakeup from Serial
> 14:03:24.074 -> 1 wake up
> 14:03:24.107 -> a
> 14:03:24.107 -> 
> 14:03:25.624 -> 2 wake up
> 14:03:25.657 -> a
> 14:03:25.657 -> 
> 14:03:29.772 -> 3 wake up
> 14:03:29.803 -> abcdefg
> 14:03:29.803 -> 
> 14:04:50.958 -> 4 wake up
> 14:04:50.958 -> test
> 14:04:50.991 -> 
> 14:05:44.378 -> 5 wake up
> 14:05:44.378 -> 0123456789
> 14:05:44.378 ->