stm32duino/STM32LowPower #123: LowPower.deepSleep(); does not enter sleep mode properly unless I call HAL_SuspendTick(); beforehand. Is this expected behavior, or am I missing something?

Toy-Jie, 2025-05-28:
`#include "STM32LowPower.h"
volatile int repetitions = 1;
const int pin = PA0;
unsigned long LastTime;

void setup()
{
    pinMode(PC13, OUTPUT);
    pinMode(pin, INPUT_PULLUP);
    LowPower.begin();
    LowPower.attachInterruptWakeup(pin, repetitionsIncrease, RISING, SHUTDOWN_MODE);
    LastTime = millis();
}

void loop()
{
    for (int i = 0; i < repetitions; i++)
    {
        digitalWrite(PC13, HIGH);
        delay(20);
        digitalWrite(PC13, LOW);
        delay(20);
    }
    if (millis() - LastTime > 3000)
    {
        LastTime = millis();
        digitalWrite(PC13, LOW);
        HAL_SuspendTick();
        LowPower.deepSleep();
    }
}

void repetitionsIncrease()
{
    repetitions = 5;
}`

===========================
Hello,
I'm having an issue where LowPower.deepSleep(); does not enter sleep mode properly unless I call HAL_SuspendTick(); beforehand.
Is this expected behavior, or am I missing something?
Thank you!
===========================
Stm32F103C8T6

fpistm, 2025-05-28:
Hi @Toy-Jie 
Is it a legacy STM32F103C8?
Which board you used?
Core version? Options?

Toy-Jie, 2025-05-29:
I'm using the STM32F103C8T6 BluePill board (definitely not GD32).
STM32 platform version: STMicroelectronics v2.9.0
Please see the screenshot below for my configuration.
Looking forward to your response. Thanks in advance!

[image]

fpistm, 2025-06-02:
I've tested and it works as expected. The deepSleep disable the irq so the tick is on more incrementented.
Seems you don't use Arduino IDE based on your screenshot so probably an issue with this environnement or it is not a legacy mcu even if you have the ST logo and name on the chips.

closed 2025-06-02 by fpistm