stm32duino/STM32LowPower #124: TimedWakeUp and AlarmWakeUp very inaccurate

PerennialNovice, 2025-06-25:
I am playing around wit the LowPower examples on Nucleo-L073RZ and fin the timings to be quite off.

When using AlarmWakeUp every 10 seconds, and printing only a single dot (with serial.flush() to ensure it gets printed) the next wakeup happens about 500ms to early.

Using TimedWakeUp and feeding it with 10.000 ms, I get approximately the same difference.

When I use AlarmWakeUp every minute, its off by about 3 seconds!
I was planning to have the system wake once a day, If this is linear, it will be of more than an hour on every wake. After three weeks it will be one day off

Is there a possibility to adjust this?

fpistm, 2025-06-25:
Hi @PerennialNovice 
Did you used LSE as RTC clock source?
By default it is the LSI which is not accurate.

See STM32RTC [README](https://github.com/stm32duino/STM32RTC?tab=readme-ov-file#api):

_RTC clock source_
* **`Source_Clock getClockSource(void)`** : get current clock source.
* **`void setClockSource(Source_Clock source, uint32_t predivA, uint32_t predivS)`** : set the clock source (`LSI_CLOCK`, `LSE_CLOCK` or `HSE_CLOCK`) and (a)synchronous prescaler values. This function must be called before `begin()`. Use `(PREDIVA_MAX + 1)` and `(PREDIVS_MAX +1)` to reset value and use computed ones. Those values have to match the following conditions: **_1Hz = RTC CLK source / ((predivA + 1) * (predivS + 1))_**

_RTC Asynchronous and Synchronous prescaler_
* **`void getPrediv(uint32_t *predivA, uint32_t *predivS)`** : get (a)synchronous prescaler values if set else computed ones for the current clock source.
* **`void setPrediv(uint32_t predivA, uint32_t predivS)`** : set (a)synchronous prescaler values.  This function must be called before `begin()`. Use `(PREDIVA_MAX + 1)` and `(PREDIVS_MAX +1)` to reset value and use computed ones. Those values have to match the following conditions: **_1Hz = RTC CLK source / ((predivA + 1) * (predivS + 1))_**

fpistm, 2025-06-26:
> Since the Nucleo-L073 has an LSE oscillator installed, I supposed that the RTC would automatically be configured to use it ...

Even if the Nucleo have a LSE, user can remove it to save power or to be able to use the related GPIO,...
At some point core and library can't known what is exactly the hardware while LSI is always prese,t. It is up to the end user to properly configure the library depending of their needs.

About your code this is not accurate, you store millis then do a print then called the low power using previous millis and current. Do not forget the low power take some times to enter (configuring the mode, configuring the rtc under the hood) then it takes some times to exit a low power mode. And also the time for the loop (exit, do some stuff in the main then enter the loop...
All those stuff took some times.

I guess you will get better accuracy using the RTC time and also using the RTC Alarm.

closed 2025-06-26 by fpistm