stm32duino/STM32LowPower #126: LowPower.shutdown(1000); does not work as expected

alexceltare2, 2025-07-07:
**Describe the bug**
The MCU should wake up after the RTC Timer interrupt tick but it doesn't. LED stays off on this mode. Other modes like LowPower.idle(1000), LowPower.sleep(1000), LowPower.deepSleep(1000) work as intended.

Steps to reproduce the behavior:

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.shutdown(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.shutdown(1000);
}

**Expected behavior**
LED Blinks every 1000ms.

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 2.3.5
 - STM32 core version: 2.10.1
 - Library version: latest
 - Tools menu settings if not the default: 
 - Upload method: Mass Storage

**Hardware (please complete the following information):**
 - Board Name: Nucleo U385RG-Q

fpistm, 2025-07-07:
Hi @alexceltare2 
This is normal the LED never blink. When you exit the shutdown mode, the board simply reset (restart).

alexceltare2, 2025-07-07:
But even this didn't work:

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.shutdown(1000);
}

Assuming LowPower.shutdown(1000); will put the CPU into sleep for 1 second then reset.

alexceltare2, 2025-07-07:
Yes, I've tested it with my U3 and works well now. Thanks!

closed 2025-07-07 by alexceltare2