stm32duino/STM32LowPower #128: sleep/deepsleep not working on Nucleo L053R8

robshep, 2025-07-15:
**Describe the bug**
LowPower sleep and deepSleep not delaying application code at all.  

**To Reproduce**
Examples > TimedWakeup

Steps to reproduce the behavior:
1. Compile
2. Upload

**Expected behavior**
Expecting these low power modes to delay application code. 
TimedWakeup uses lowpower deep sleep to flash the LED at 1s/1s duty, but the LED just flickers as fast as possible with no delay.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Arduino IDE version: 2.3.6
 - STM32 core version: 2.10.1
 - Library version: 1.4.0  (RTC @ 1.7.0)
 - Tools menu settings if not the default: 
 - Upload method: STLink over SWD

**Hardware (please complete the following information):**
 - Board option: Nucleo 64
 - Board Name: Nucleo L053R8
 - Hardware Revision: MB1136 Rev C
 - Extra hardware used if any: None

**Additional context**
Stock out of the box Nucleo, clean Arduino env, clean library env.

fpistm, 2025-07-16:
Hi @robshep 
I've tested with a Nucleo L053R8 and it works as expected.
Ok it is not exactly 1000 ms by default but several factor have to be take in account:
* RTC alarm is configured before entering in deep sleep (STOP2).
* RTC clock source precision. By default LSI is used.

[image]

[image]

* Wakeup time (including extra hal delay of 10 ms for the stability).

## LSI as RTC source clock (default):

[image]

## LSE as RTC source clock:

```diff
#include "STM32LowPower.h"
+ STM32RTC& rtc = STM32RTC::getInstance();

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
+  rtc.setClockSource(STM32RTC::LSE_CLOCK);
```

[image]

So I see no issue here about deepSleep not working.

robshep, 2025-07-17:
Thank you for the investigation.

Today on fresh power - it works as expected.  
At least 3 hours spent trying to get it to work.  
Between my original report and today, an update to Cube brought a an update to the STLink portion of the Nucleo.  
This is the only thing to have changed.  Puzzled.