stm32duino/STM32LowPower #129: STM32U073CCT6 does not wake up reliably from deepSleep() without sleep()/delay() of at least ~195us

ndoo, 2025-07-16:
**Describe the bug**

STM32U0 (at least, STM32U073CCT6) does not execute properly without some kind of delay after calling `deepSleep()`.

Blink sketch below blinks erratically unless `STM32_WAKEUP_LATENCY_USEC` is >= 195 (may vary? not sure what an appropriate safety margin is).

**To Reproduce**

```
#include <STM32LowPower.h>
#include <STM32RTC.h>

#define LED_GREEN PA6
#define STM32_WAKEUP_LATENCY_USEC 195
#include "STM32LowPower.h"

void setup() {
  STM32RTC& rtc = STM32RTC::getInstance();
  rtc.setClockSource(STM32RTC::LSE_CLOCK);
  rtc.begin();

  pinMode(LED_GREEN, OUTPUT);

  // Configure low power
  LowPower.begin();
}
 
void loop() {
  digitalWrite(LED_GREEN, HIGH);
  LowPower.deepSleep(1000);
  LowPower.sleep(STM32_WAKEUP_LATENCY_USEC);
  digitalWrite(LED_GREEN, LOW);
  LowPower.deepSleep(1000);
  LowPower.sleep(STM32_WAKEUP_LATENCY_USEC);
}
```

**Expected behavior**
1000ms + (wakeup time) blink interval

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS
 - Arduino IDE version: 2.3.6
 - STM32 core version: 2.10.1
 - Library version: 1.4.0 (RTC: 1.7.0)
 - Tools menu settings if not the default: As default + CDC serial (sketch does not use it)
 - Upload method: USB DFU

**Hardware (please complete the following information):**
 - Board Name: Custom STM32U073CCT6 PCB
 - Hardware Revision: N/A
 - Extra hardware used if any: Green LED

**Additional context**

Perhaps a larger `HAL_Delay()` is needed for STM32U0.

ndoo, 2025-07-16:
I tested using these in my sketch:

```c
#define STM32_WAKEUP_LATENCY_USEC 189
..
HAL_Delay(STM32_WAKEUP_LATENCY_USEC);
```

It seems to execute subsequent code reliably, using 188us fails.

This is using LSE clock; of course, some buffer should be built in, I think STMicro would know better a safe value to use.

fpistm, 2025-07-16:
Hi,
seems very strange to call a sleep before a deepsleep.
Anyway, deepsleep requires some time to enter and exit so a delay of 195µs is not very surprising.
Typical time wakup fro stop2 mode is around 26-29µs but some to this time you can add the time to program the RTC before to enter in stop2 mode, the time to enter in this mode, then the config after the wakeup (clock config, RAM retention,...)

ndoo, 2025-07-16:
Hi @fpistm ,

I have changed it to `HAL_Delay()` of 250 just to have a good safety margin because the deployed MCU will be in a remote area.

HAL_Delay 10 is not enough for STM32U0, it fails consistently if `HAL_Delay()` of 188s or less after `deepSleep()` exits.

Perhaps this should be documented somewhere..? We kind of found it by trial and error, after wondering why a bunch of GPIO and serial stuff was failing unpredictably.

ndoo, 2025-07-16:
Or... should this be factored in to the library? i.e. some per-variant delay within the linked code
https://github.com/stm32duino/STM32LowPower/blob/471587035bfebb163710f477c174d622d8ef8077/src/low_power.c#L800

fpistm, 2025-07-16:
I've tested with a Nucleo U083RC and it works as expected:

[image]

What is your hardware exactly with a STM32U073CC? And how you test it as it is not yet supported in the STM32 core ?

ndoo, 2025-07-18:
> I've tested with a Nucleo U083RC and it works as expected:
> 
> <img alt="Image" width="2000" height="1705" src="https://private-user-images.githubusercontent.com/20641798/467019619-15275e0d-14fa-43a5-9dec-90c0dfcb1812.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI4MDUwNzksIm5iZiI6MTc1MjgwNDc3OSwicGF0aCI6Ii8yMDY0MTc5OC80NjcwMTk2MTktMTUyNzVlMGQtMTRmYS00M2E1LTlkZWMtOTBjMGRmY2IxODEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE4VDAyMTI1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ5Y2YyNGFjODIwMDFmYzI0NDE4YzZmZTVjMzA3ZGU0NTlhMGIxZTVjY2Y5YzI4MzQ1YjRmMTBhMTkwYWZlMDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JB1G-KIHNBqRsy1okpfo3ALBHpeiCiazrnI_kt1zx2w">
> What is your hardware exactly with a STM32U073CC? And how you test it as it is not yet supported in the STM32 core ?

Hi,

Is STM32U073CCTx not supported? I am using STM32U073RCTx and it builds and runs fine, with the exception of cases like this.